moonlight-stream/moonlight-embedded #342: No keyboard and mouse when using -input command line switch

substring, 2016-03-17:
Please provide the following info.

**_NVidia Geforce Experience version:_** 2.10.4
**Moonlight Embedded version:** 2.1.4
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._ [Recalbox](http://www.recalbox.com) or see the recalbox project on github

**Output of Moonlight Embedded:** don't have it at hand for now

**What is the expected result?**
When adding -input and -map for specific pads configuration, the keyboard and mouse should be added anyway

**What happens instead of that?**
the keyboard and mouse won't be added if a -input is passed as a moonlight argument

wouterverduin, 2016-03-25:
How do you properly pass an -input signal?

irtimmer, 2016-04-03:
You have to separately provide -input options for the mouse and keyboard.

closed 2016-04-03 by irtimmer