moonlight-stream/moonlight-embedded #343: Audio lag of 0.3-0.4 seconds on Raspberry Pi 2

fadetonoir, 2016-03-18:
I've been trying to get moonlight-embedded setup to stream games from my desktop PC to my TV via a Raspberry Pi for a couple of months now but I keep having issues with audio. I did have an original Raspberry Pi Model B and with that the audio stuttered constantly, making the games unplayable. At Christmas I upgraded to a Raspberry Pi 2 and thought I would try Moonlight again. The audio now plays clearly, but it lags behind the image with a noticeable delay of about 0.3-0.4 seconds. This happens with default Moonlight streaming settings and also at 1080p.

Hope someone can help me out with this. Happy to answer questions about my setup and if there are any settings you think I should try, please let me know.

The desktop PC had a clean installation of Windows 10 last week, then installed the latest nVidia drivers and GeForce Experience. The Raspberry Pi 2 has a clean installation of NOOBS with Moonlight-Embedded and nothing else installed or running. Both are connected through a router via ethernet. I usually start the streaming using JuiceSSH on my phone but I used Putty on my laptop for this test to get the output I've pasted below (with the same results).

Info requested - 

NVidia Geforce Experience version: 2.10.2.40

Moonlight Embedded version: 2.1.4

Moonlight Embedded running on: Raspberry Pi 2

Output of Moonlight Embedded:
pi@raspberrypi:~ $ moonlight stream -1080
Moonlight Embedded 2.1.4 (EMBEDDED;PI)
Connect to 192.168.0.11...
Stream 1920 x 1080, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Alsa shortm write, write 12 frames
Discarding RTP packet queued for too long
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Loss Stats: Transaction failed: 32
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Terminating RTSP handshake...done
Cleaning up platform...done
pi@raspberrypi:~ $ 

What is the expected result? Audio perfectly in sync

What happens instead of that? Audio lags by about 1/3 - 1/2 second

dead, 2016-03-18:
Maybe it's your network... Did you try with 720?

fadetonoir, 2016-03-18:
Yes, I said I tried it at default settings, which is 720p, with the same results.

How can I test to see if it's my network? As I said, I've got PC and Pi connected via ethernet and my network is stable and fast enough most of the time - I can certainly use Steam streaming from PC to laptop with no lag.

Unknownforce351, 2016-03-24:
What kind of TV do you have?  Is the PI connected directly to TV or through an A/V Receiver or audio passthrough TV to Receiver?

irtimmer, 2016-03-27:
Duplicate of #280

closed 2016-03-27 by irtimmer