moonlight-stream/moonlight-embedded #347: Cannot change audio output device

Trondomer, 2016-03-26:
I have a USB soundcard connected to my RPI2 and I'm trying to make the audio stream from moonlight go through it instead of going over HDMI to my TV speakers. For some reason the system default for audio is HDMI.

My output for cat /proc/asound/cards is:

0 [ALSA ]: bcm2835 - bcm2835 ALSA
bcm2835 ALSA
1 [Device ]: USB-Audio - USB PnP Sound Device
USB PnP Sound Device at usb-3f980000.usb-1.4, full speed

What exactly am I supposed to enter in the -audio option? Whatever I've tried I get Unknown PCM. To be more precise, if I enter -audio 1 I get: 

Starting audio stream...ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM 1

Also, I'm kinda new to everything linux so please bear with me on this one. I'm using the Luna launcher on OSMC and I assume if I were to change the "#audio = sysdefault" line in the config file that every stream started via Luna would output audio through the selected device. Now, where the heck do I locate the config file? Or do I have to create one?

Please provide the following info.

_NVidia Geforce Experience version: 2.10.2.40
Moonlight Embedded version:2.1.4
Moonlight Embedded running on: Raspberry Pi 2 B

dead, 2016-03-26:
I think this can help you: https://github.com/dead/moonlight-openelec-rpi2/issues/6#issuecomment-192888433

Trondomer, 2016-03-27:
Thank you! 
Adding the parameter hw:1,0 made the audio go through the USB soundcard while hw:0,0 seems to be the default ALSA device.

I'm curious, can you explain what exactly does hw mean? And what does 1,0 point to? I'm guessing 1 references to "1 [Device ]: USB-Audio", but what about the zero after to comma.

closed 2016-03-27 by irtimmer

dead, 2016-03-27:
I think 1 is the card and 0 is the device, not sure. :smile:

tzani, 2017-06-27:
$  moonlight stream 192.68.0.15 -audio hw:1,0
With the 1 and the 0 corresponding to the **Card 1** and **Device 0** in:

$ aplay -L

**card 1**: AUDIO [USB  AUDIO], **device 0**: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

At least as far as I understand it.