moonlight-stream/moonlight-embedded #349: 404 Error Moonlight-Embedded

mclucario, 2016-03-29:
Please provide the following info.

NVidia Geforce Experience version: 2.11.2.46
Moonlight Embedded version: 2.1.4 (EMBEDDED;PI)
Moonlight Embedded running on:** Raspberry Pi 2016-03-18-raspbian-jessie

Output of Moonlight Embedded:

pi@raspberrypi:~ $ moonlight stream 192.168.0.14
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...RTSP OPTIONS request failed: 404
failed: 404

What is the expected result?:

The Stream to actually start.

What happens instead of that?:

A weird 404 error for some reason.

dead, 2016-03-29:
https://github.com/moonlight-stream/moonlight-common-c/issues/16

mclucario, 2016-03-29:
@dead Thank you. I didn't see that. I hope the update gets pushes soon. Is there a version I can compile and use from the releases which work?

cgutman, 2016-03-31:
The latest moonlight-common-c has support for GFE 2.11.2.46

mclucario, 2016-03-31:
@cgutman Do I just compile normally with the moonlight-embedded-2.1.4.tar.xz or do I need to do something else? I wanna compile on my raspberry pi if possible.

dead, 2016-04-01:
@mclucario You can try compiling this: https://github.com/dead/moonlight-embedded

ronlaws86, 2016-04-01:
Any ideas when this will be fixed? Some of us are getting withdrawal symptoms!

mclucario, 2016-04-02:
@dead I created a clone with git but couldn't compile it on my RPI. I missing libaries but I can find them in the Apt-Get nor on google.

7hunderbug, 2016-04-02:
@mclucario, you can use irtimmer's repo now, I've tested it and it works.
Clone irtimmer's repo and follow these instructions: https://github.com/irtimmer/moonlight-embedded/wiki/Compilation
Pay attention to required libraries for compilation at the bottom of that page. 
If compiling on Raspbian, ignore the instructions to use raspbian/jessie branch and ignore instructions to use `fakeroot debian/rules binary` to compile. Instead follow the "cmake" way:
`git submodule update --init`
`mkdir build`
`cd build`
`cmake ../`
`make`
`sudo make install`

7hunderbug, 2016-04-02:
Ah...., I forgot to mention that if you're using Raspbian to compile, you do need the two small patches to the CMakeLists.txt files (the one in the source tree root directory and the one in libgamestream/) to allow for the pre-packaged version of cmake (3.0.2) on Raspbian and to change the C compile flags to -std=gnu99

mclucario, 2016-04-02:
@7hunderbug Im using a compiles version of cmake-3.5.1, do I need to change it anyway?

7hunderbug, 2016-04-02:
@mclucario you won't need the changes for cmake but you might need the changes to the C_FLAGS if compiling on Raspbian. You'll know if, when you try to compile, you get errors relating to C standards.

If you get errors and the build fails, change this line:
`set_property(TARGET moonlight PROPERTY C_STANDARD 99)`
to this:
`set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=gnu99")`
in CMakeLists.txt and in libgamestream/CMakeLists.txt.

mclucario, 2016-04-02:
@7hunderbug Oh I see now. I dont actually need to do that. I compiled moonlight 5 times already following the instructions of this website: http://jordanhotmann.com/guide-install-moonlight-on-raspbianretropie/

irtimmer, 2016-04-03:
Fixes for GFE 2.11 have landed in Moonlight Embedded

closed 2016-04-03 by irtimmer