moonlight-stream/moonlight-embedded #350: Platform 'default' not found

amylizzle, 2016-03-30:
After compiling from source (in order to get GFE 2.11 compatibility), all moonlight commands fail with the message "Platform 'default' not found".

Please provide the following info.

**_NVidia Geforce Experience version:_** 2.11.2.46
**Moonlight Embedded version:** 2.1.4 (built from source from git as of 30/03/16 18:00 GMT)
**Moonlight Embedded running on:** _Raspberry Pi_

**Output of Moonlight Embedded:** 
Moonlight Embedded 2.1.4 (EMBEDDED;PI)
Platform 'default' not found

**What is the expected result?** Pairing, streaming, all commands fail except help.

**What happens instead of that?** Error message "Platform 'default' not found, followed by immediate termination of moonlight.

**Build Log**

pi@raspberrypi ~/moonlight/moonlight-embedded/build $ cmake ../
-- The C compiler identification is GNU 4.7.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ALSA: /usr/lib/arm-linux-gnueabihf/libasound.so (found version "1.0.25")
-- Found Opus: /usr/lib/libopus.so
-- Found Broadcom: /opt/vc/include
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Could NOT find Amlogic (missing:  AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26")
-- checking for module 'libevdev'
--   found libevdev, version 1.4.6
-- checking for module 'libudev'
--   found libudev, version 175
-- checking for module 'sdl2>=2.0.4'
--   package 'sdl2>=2.0.4' not found
-- checking for module 'libavcodec'
--   package 'libavcodec' not found
-- checking for module 'libavutil'
--   package 'libavutil' not found
-- checking for module 'x11-xcb'
--   package 'x11-xcb' not found
-- checking for module 'vdpau'
--   package 'vdpau' not found
-- checking for module 'libpulse-simple'
--   package 'libpulse-simple' not found
-- checking for module 'libcec>=3.0.0'
--   package 'libcec>=3.0.0' not found
-- checking for one of the modules 'uuid'
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.26.0")
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found version "1.0.1e")
-- Found EXPAT: /usr/lib/arm-linux-gnueabihf/libexpat.so (found version "2.1.0")
-- checking for module 'avahi-client'
--   found avahi-client, version 0.6.31
-- checking for module 'libenet'
--   found libenet, version 1.3.13
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/moonlight/moonlight-embedded/build
[... 68 lines cut ...]
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/libmoonlight-pi.so
-- Removed runtime path from "/usr/local/lib/libmoonlight-pi.so"
-- Installing: /usr/local/bin/moonlight
-- Removed runtime path from "/usr/local/bin/moonlight"
-- Up-to-date: /usr/local/share/moonlight/mappings
-- Installing: /usr/local/share/moonlight/mappings/xbox360.conf
-- Installing: /usr/local/share/moonlight/mappings/rumblepad2.conf
-- Installing: /usr/local/share/moonlight/mappings/dualshock3.conf
-- Installing: /usr/local/share/moonlight/mappings/default.conf
-- Installing: /usr/local/share/moonlight/mappings/dualshock4.conf
-- Installing: /usr/local/etc/moonlight.conf
-- Installing: /usr/local/lib/libgamestream.so.2.1.4
-- Up-to-date: /usr/local/lib/libgamestream.so.0
-- Up-to-date: /usr/local/lib/libgamestream.so
-- Removed runtime path from "/usr/local/lib/libgamestream.so.2.1.4"
-- Installing: /usr/local/lib/libmoonlight-common.so.2.1.4
-- Up-to-date: /usr/local/lib/libmoonlight-common.so.0
-- Up-to-date: /usr/local/lib/libmoonlight-common.so
-- Installing: /usr/local/share/man/man1/moonlight.1

pi@raspberrypi ~/moonlight/moonlight-embedded/build $ moonlight pair 192.168.200.7
Moonlight Embedded 2.1.4 (EMBEDDED;PI)
Platform 'default' not found

amylizzle, 2016-03-30:
So I'm pretty sure I know what's going on, I'm just not 100% sure why.

It looks like `platform_check()` in platform.c is being compiled to a `return 0` stub, possibly because HAVE_PI and the like aren't getting defined properly (this may be the fault of the build process, don't understand it well enough to be sure). 

I managed to hack around the problem by commenting out `if` statements such that the code that *should* run if HAVE_PI is defined always runs, though of course this is not a general solution.

amylizzle, 2016-03-31:
Errorcode -1 seems to be caused by a protocol change in the streaming API. In `libgamestream/client.c`, moonlight searches the returned XML for a node `cancel` which is not there in the latest protocol, and so fails. Commenting out that search gets me to the next error which was reported in #349 and is also a protocol issue.

I'm not sure what the ideal fix for #350 is, but it seems to be an error in the build process that could be resolved by requiring libavcodec if Broadcom is detected.

dead, 2016-03-31:
Moonlight-embedding is not working with GFE 2.11.2.46 yet. To fix the Errorcode starting app: -1 I had to revert this commit (https://github.com/irtimmer/moonlight-embedded/commit/016ac91c1c19f868cc46cac5b4d794b414b8d74e).

About the HAVE_PI maybe it's a problem with CMake, but it's working fine here...

Unknownforce351, 2016-03-31:
Edit platform.c and add the following in where you find it:
```
  #ifdef HAVE_PI
  if (std || strcmp(name, "pi") == 0) {
    void *handle = dlopen("libmoonlight-pi.so", RTLD_NOW | RTLD_GLOBAL);
	if (handle == NULL) fprintf(stderr, "%s\n", dlerror());  // <------ ADD THIS LINE
    if (handle != NULL && dlsym(RTLD_DEFAULT, "bcm_host_init") != NULL)
      return PI;
  }
  #endif
```

amylizzle, 2016-03-31:
CMake was compiled from source, version 3.2.3. Could still be something to do with my setup, this pi isn't exactly a clean dev machine.

I was actually playing a little with that code already, and it always got a proper handle. I'm fairly sure it was just a build issue.

Unknownforce351, 2016-03-31:
Ah!  I just remember running into #350 when debugging the OMX audio code I recently PR'd and was committed. I used that to debug and found that it wouldn't get a handle because of some bad declaration/definitions in my code.  I was just making sure it wasn't related to that code I pushed.  Didn't want to be the cause of problems!

cgutman, 2016-03-31:
Moonlight-common-c is updated upstream to support GFE 2.11.2.46.

amylizzle, 2016-04-01:
Well, my build environment is still screwy apparently.

After running `cmake ../` in `build/`, running `make` 

>pi@raspberrypi ~/moonlight/moonlight-embedded/build $ make
>Scanning dependencies of target moonlight-common
>make[2]: *** No rule to make target 'libgamestream/CMakeFiles/moonlight-common.dir/build'.  >Stop.
>CMakeFiles/Makefile2:184: recipe for target 'libgamestream/CMakeFiles/moonlight-common.dir/all' >failed
>make[1]: *** [libgamestream/CMakeFiles/moonlight-common.dir/all] Error 2
>Makefile:116: recipe for target 'all' failed
>make: *** [all] Error 2

7hunderbug, 2016-04-01:
Using a clean build environment and current git master as of today results in a working build (for me) on RPi Jessie. I've used irtimmer's moonlight-common-c submodule and the packaged libenet-dev from Jessie repos (rather than cgutman's one in his moonlight-common-c repo).

I have seen the "Platform 'default' not found" error when calling the compiled moonlight executable from my home folder (e.g. `./sources/moonlight-embedded/build/moonlight stream`). However, this error didn't occur when running moonlight from within the build directory (`cd ~/sources/moonlight-embedded/build/ & ./moonlight stream`) or when running moonlight from the system install after running `sudo make install`. I think the error is related to the location of the `libmoonlight-pi.so` compiled library that is compiled along with the main app.

amylizzle, 2016-04-02:
Did a clean build, everything is working now! Thanks for all your help :)

I'm happy to mark this as resolved? Seems to have just been an issue on my end, excepting the protocol stuff of course.

closed 2016-04-02 by irtimmer