moonlight-stream/moonlight-embedded #353: RTSP handshake failed 404 & EVIOGRAB failed w. error 16

klen-ghost, 2016-04-01:
Please provide the following info.

**_NVidia Geforce Experience version:_2.11.2.55
**Moonlight Embedded version:** 2.1.4
**Moonlight Embedded running on:** _Raspberry Pi 2
**Output of Moonlight Embedded:**
Moonlight embedded 2.1.4 (EMBEDDED:pulse;p#)
connect to 192.168.2.4
stream 1280 x 720, 60fps, 10000kbps
initializing platform...done
resolving host name...done
Starting RTSP handshake...RTSP OPTIONS request failed:404
failed:404
EVIOGRAB failed with error 16

**What is the expected result?**
moonlight/steam in BPM on rpi2 / TV 

**What happens instead of that?**
after trying to start this error comes on the tv / rpi2 output.
however steam bpm is up&running on the host pc!

7hunderbug, 2016-04-01:
Duplicate issue. See here: https://github.com/irtimmer/moonlight-embedded/issues/349

You'll either have to compile moonlight from git to enable support for GFE 2.11.2.55, (the precompiled binary doesn't work with this version) or downgrade GFE to version 2.10.2.40.

GFE 2.10.2.40 available here: [http://uk.download.nvidia.com/GFE/GFEClient/2.10.2.40/GeForce_Experience_v2.10.2.40.exe](http://uk.download.nvidia.com/GFE/GFEClient/2.10.2.40/GeForce_Experience_v2.10.2.40.exe)

closed 2016-04-01 by irtimmer

klen-ghost, 2016-04-02:
Thanks. I actually installed a gfe 2.9.#.# version. On first run however
gfe sadly auto update.. did not know moonlight did not support this version.
Installed

only problem is that GFE keeps auto updating itself.
made a block on gfe.exe en launchgfe.exe in firewall
edited host file added following :

# 127.0.0.1 services.gfe.nvidia.com

# 127.0.0.1 https://services.gfe.nvidia.com/GFE/getClientUpdate/

# 127.0.0.1 https://services.gfe.nvidia.com/GFE/getClientUpdate

# 127.0.0.1
http://international.download.nvidia.com/GFE/GFEClient/2.11.2.55/GeForce_Experience_Update_v2.11.2.55.exe

127.0.0.1 https://services.gfe.nvidia.com/GFE/getClientUpdate/

127.0.0.1 https://services.gfe.nvidia.com/GFE/getClientUpdate

127.0.0.1
http://international.download.nvidia.com/GFE/GFEClient/2.11.2.55/GeForce_Experience_Update_v2.11.2.55.exe

127.0.0.1 services.gfe.nvidia.com

can't seem to stop this stupid auto update. it starts gfe and starts
updating right after.
maybe you know how to stop this auto update BS.

thnx!

best regards!

7hunderbug, 2016-04-02:
The hosts file takes only FQDN or hostnames, not URLs. So the only line you need is:
`127.0.0.1 services.gfe.nvidia.com`

This works for me.

If it doesn't work for you, it is probably because you computer already has the updated GFE installer cached somewhere, perhaps in temp folder or similar? Try uninstalling GFE, then running CCleaner or similar to delete temporary and obsolete files, then installing GFE again.

klen-ghost, 2016-04-02:
I'll try deleting temp & cache of the pc that will probably do the trick.
Is there a command to "un" pair a device (moonlight said already paired,
but since new gfe install it is not paired anymore, resulting in a
situation that i can't pair.

Thanks and have a great weekend ;)

7hunderbug, 2016-04-02:
Yes, there is a "Forget Devices" button in GFE under the "Shield" preferences item. Have a good weekend yourself!

klen-ghost, 2016-04-02:
Thanks!
New issue now steam bpm is shown on rpi2 /moonlight.
Controller input is not working. Only left and right are working.
Inside retropie everything does work.
Already tried :
Reconfiguring controller in retropie (detected as shawin clone) ps3 clone
via a 2.4ghz connection)
After that i tried this:
Sudo ./retropie_setup.sh
Option 3
Then searched for ps3 controller and enabled/installed it.
Still not working.

It was working on a different os (recalbox with moonlight addon)
But that is based on openelec and my kodi wasn't stable.

Thanks
Best regards

7hunderbug, 2016-04-02:
Have you tried making a moonlight input mapping file for your controller yet?
 
e.g. make a mapping file like this: `moonlight map -mapping ~/ps3clone.map -input /dev/input/event0`
use mapping file like this: `moonlight stream -mapping ~/ps3clone.map -input /dev/input/event0 game.stream.pc.ip.address`

You'll need to find the correct input event handle, it could be /dev/input/event0 or /dev/input/event1 etc. The `evtest` tool is handy for finding this and for testing your controller.