moonlight-stream/moonlight-embedded #355: [2.2.0] libenet1a is not installable

sjmorrow, 2016-04-03:
When I try to install the latest version of moonlight-embedded from `deb http://archive.itimmer.nl/raspbian/moonlight wheezy main` I get the following error:

```
The following packages have unmet dependencies:
 moonlight-embedded : Depends: libenet1a but it is not installable
E: Unable to correct problems, you have held broken packages.
Please provide the following info.
```

**_NVidia Geforce Experience version:_** 2.11.2.55
**Moonlight Embedded version:** 2.2.0
**Moonlight Embedded running on:** _Raspberry Pi_ with RetroPie 3.6

alfalfa1978, 2016-04-03:
I downloaded libenet1a and libenet-dev here:
https://packages.debian.org/de/wheezy/armhf/libenet1a/download
https://packages.debian.org/wheezy/armhf/libenet-dev/download

but when I sudo cmake now, it aborts compiling at 67 % with the error: 

libgamestream/libmoonlight-common.so.2.2.0: undefined reference to `enet_peer_timeout'
collect2: error: ld returned 1 exit status
CMakeFiles/moonlight.dir/build.make:440: recipe for target 'moonlight' failed
make[2]: *** [moonlight] Error 1
CMakeFiles/Makefile2:106: recipe for target 'CMakeFiles/moonlight.dir/all' failed
make[1]: *** [CMakeFiles/moonlight.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Also NV GE 2.11.2.55, Moonlight 2.2.0 and RPi 3 with RetroPie 3.6

7hunderbug, 2016-04-03:
Isn't Retropie 3.6 based of Raspbian Jessie now? If so you'll need the Jessie apt repo, not the Wheezy one.

i.e. `deb http://archive.itimmer.nl/raspbian/moonlight jessie main`
not `deb http://archive.itimmer.nl/raspbian/moonlight wheezy main`

sjmorrow, 2016-04-03:
@7hunderbug has saved the day. You are correct, that fixed the issue.

For anyone coming to this issue from a google search, the fix is as follows:

1. Open terminal / ssh to your device
2. `sudo nano /etc/apt/sources.list`
3. Change `deb http://archive.itimmer.nl/raspbian/moonlight wheezy main` to `deb http://archive.itimmer.nl/raspbian/moonlight jessie main`
4. Ctrl + O to write changes to disk, press ENTER to confirm
5. Ctrl + X to exit nano
6. `sudo apt-get update`
7. `sudo apt-get install moonlight-embedded`

closed 2016-04-03 by sjmorrow