moonlight-stream/moonlight-embedded #356: Surround audio distortion with 2.10 or 2.11

Unknownforce351, 2016-04-04:
**_NVidia Geforce Experience version:_**
2.10.X or 2.11.X
**Moonlight Embedded version:**
2.2.0, but pretty much any version with surround.
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
RPi2 - Raspbian Jessie
**Output of Moonlight Embedded:**
Normal
**What is the expected result?**
Clear Surround Audio
**What happens instead of that?**
Distorted audio, Audio sounds like it's "sped up" and is heavily distorted and crackling.

Strange thing on this is... 2.9 works just fine.  2.10, 2.11 cause the audio distortion.
Happens with any audio output, OMX or ALSA, but only with -surround option, which I find strange.  Did they make some changes to surround in GFE, like add 7.1 or something?  I seem to remember getting similar issues like this when testing the OMX code that I recently PR'd, when the buffer size was set incorrectly.  Maybe there's buffer size changes to audio in gen 5+?

I've tried the usual stuff, rebooting, reloading drivers, using BETA...
I tried also reloading my Pi with fresh Jessie and moonlight install, same result.

I don't know if it's actually a moonlight thing or maybe something with 2.10/2.11?
As I said, I'm fine using 2.9, but would like to use the latest version for the audio fixes, cause I use RDP for the most part and I do get some audio issues here and there (drops out for short periods of time)

Unknownforce351, 2016-04-11:
To add an update here... I just setup a fresh Raspbian Jessie install on a new RPi3, and still experience the same issue here.  When I get some time, I'm going to mess with some buffer sizes and dive into the common audio code and see where the problem could be here.  Something had to have changed in gen 5+ server...

cgutman, 2016-04-13:
Interesting, when I enable surround sound with Moonlight Android on my S7, it sounds fine. I don't have a real 5.1 system to confirm, but I'd expect that issues like you describe would survive the downmixing.

Unknownforce351, 2016-04-15:
Does android use the same buffer/frame sizes and channel counts?  I'm not very proficient with Java, so in the little time I looked at it, I couldn't find the audio code that defines them...

cgutman, 2016-04-15:
https://github.com/moonlight-stream/moonlight-android/blob/master/app/src/main/java/com/limelight/binding/audio/AndroidAudioRenderer.java

It's possible that there is a bug in moonlight-common-c that moonlight-common doesn't have but there's nothing that I can see.

Unknownforce351, 2016-04-15:
One noticeable difference is that RTP_RECV_BUFFER is (64 * 1024) in common, but only (64 * MAX_PACKET_SIZE) in common-c which equates to (64 * 250).  Maybe this?  I'll test when I get home.

Unknownforce351, 2016-04-18:
Well, setting it to 64 * 1024 fixed the distortion, so that explains why your sound is just fine on Android.  However, surround sound is still broken.  Strange results here:

Using OMX, all channels plays out of front left and right only.  Using a 5.1 test, front left and right are normal, center plays out of FL and FR together, but not from the actual center channel, RL comes from FL, and RR comes from FR.  Basically, it's only got 2.0 audio.  -surround option has no effect, I even tried forcing 5.1 by changing the default value to 5.1 in config.c, no change.

Now here's where it gets strange...

Using ALSA instead of OMX (-audio sysdefault), if I have surround off, it plays just like OMX does above, expected being 2.0 audio.  However, if I have surround ON, it only plays front left and front right sounds.  The center, LFE, RR and RL do not play at all.  Now, understand that this is expected because HDMI 5.1 through sysdefault ALSA on the Pi does not work due to a firmware/driver issue.

More to the point, the surround option has an effect on ALSA, but not on OMX.  This is rather puzzling.

Reverting to 2.9 resolves the problem, surround sound works as expected again. (And the 64 * 1024 change has no adverse effects that I can tell)

cgutman, 2016-04-18:
The socket buffering change only masks audio decoding performance issues. Because the Opus decoding workload has over 2x the number of channels, the audio decoding can't keep up sometimes. I would have bet that the Pi 2 could keep up (though the Pi 1 even struggles with stereo), but I guess not.

Unknownforce351, 2016-04-18:
Well it is an issue even on the Pi3.  But why would it only be a problem in 2.10 and 2.11, that's what's strange to me...  Something significant related to Surround Audio must have changed in these newer gens.  May be a stupid question here, but where do the GFE API's come from?  Or was it originally developed just using packet sniffing/decoding?

Unknownforce351, 2016-04-25:
After testing further, I found out that it's Remote Desktop that doesn't have surround sound.  If I stream games directly rather than using RDP, surround sound works properly.  I think it's GFE that's the problem, with their "fixed audio in Remote Desktop" fix recently implemented.  I'm closing this issue.

closed 2016-04-25 by Unknownforce351