moonlight-stream/moonlight-embedded #358: Pairing Loop

theFisher86, 2016-04-07:
Please provide the following info.

**NVidia Geforce Experience 2.11.2.55**
**Moonlight Embedded version:Moonlight Embedded 2.1.4 (EMBEDDED;PULSE;PI)**
**Moonlight Embedded running on: Raspberry Pi 2, RetroPie** 

Everytime I try to use moonlight it says I need to pair, even though it's already paired.  See below:

`pi@retropie:~ $ moonlight pair 192.168.0.51`
`Moonlight Embedded 2.1.4 (EMBEDDED;PULSE;PI)`
`Connect to 192.168.0.51...`
`Please enter the following PIN on the target PC: 7026`
`1017 / 1017`
`Succesfully paired`
`pi@retropie:~ $ moonlight list 192.168.0.51`
`Moonlight Embedded 2.1.4 (EMBEDDED;PULSE;PI)`
`Connect to 192.168.0.51...`
`You must pair with the PC first`

dead, 2016-04-08:
Try update to moonlight 2.2.0

closed 2016-04-08 by irtimmer

theFisher86, 2016-04-08:
I tried to update to 2.2 however I was unable to compile due to multiple errors. Is there a package. Available for 2.2 somewhere?                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 smartphone on the Verizon Wireless 4G LTE network.                                                                                                                                                                                                                From: GustavoSent: Thursday, April 7, 2016 10:09 PMTo: irtimmer/moonlight-embeddedReply To: irtimmer/moonlight-embeddedCc: Aaron FisherSubject: Re: [irtimmer/moonlight-embedded] Pairing Loop (#358)Try update to moonlight 2.2.0

dead, 2016-04-08:
There are packages available, check https://github.com/irtimmer/moonlight-embedded/wiki/Packages

theFisher86, 2016-04-08:
I'm pretty sure the package only goes up to 2.1.4  Unless there's an
apt-get command that I'm not aware of that will show newer releases
(totally possible).  I'm not a complete Linux Noob but there's a lot I
don't know!

cgutman, 2016-04-08:
You probably need to update the cached package information.

Use: sudo apt-get update

Then try: sudo apt-get upgrade

theFisher86, 2016-04-08:
Alright, I figured out the problem.  The 2.2.0 version of moonlight still *says* 2.1.4. See below:

**Install version 2.2.0-1jessie1**
```
pi@retropie:~ $ sudo apt-get install moonlight-embedded=2.2.0-1jessie1
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  moonlight-embedded
0 upgraded, 1 newly installed, 0 to remove and 52 not upgraded.
Need to get 0 B/82.9 kB of archives.
After this operation, 297 kB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  moonlight-embedded
Install these packages without verification? [y/N] y
Selecting previously unselected package moonlight-embedded.
(Reading database ... 74465 files and directories currently installed.)
Preparing to unpack .../moonlight-embedded_2.2.0-1jessie1_armhf.deb ...
Unpacking moonlight-embedded (2.2.0-1jessie1) ...
Processing triggers for man-db (2.7.0.2-5) ...
Setting up moonlight-embedded (2.2.0-1jessie1) ...
Processing triggers for libc-bin (2.19-18+deb8u3) ...
```
**Then check the version**

`pi@retropie:~ $ moonlight`
`Moonlight Embedded ` **2.1.4** ` (EMBEDDED;PULSE;PI)`
`Usage: moonlight [action] (options) [host]`

It seems version 2.2.0-1 still reports as Moonlight Embedded 2.1.4

theFisher86, 2016-04-08:
Regardless, even after all this the problem still exists.

```
pi@retropie:~ $ moonlight pair 192.168.0.51
Moonlight Embedded 2.1.4 (EMBEDDED;PULSE;PI)
Connect to 192.168.0.51...
Please enter the following PIN on the target PC: 2562
Failed to pair to server: Already paired
pi@retropie:~ $ moonlight list 192.168.0.51
Moonlight Embedded 2.1.4 (EMBEDDED;PULSE;PI)
Connect to 192.168.0.51...
You must pair with the PC first
```

irtimmer, 2016-04-08:
Version 2.2.0 doesn't still says 2.1.4. It says 2.1.4 because you are still running 2.1.4 and therefore the problem still exists because 2.1.4 doesn't support GFE 2.11

theFisher86, 2016-04-08:
Well I posted the exact install process and it shows it's installing 2.2.0:

```
Preparing to unpack .../moonlight-embedded_2.2.0-1jessie1_armhf.deb ...
Unpacking moonlight-embedded (2.2.0-1jessie1) ...
Processing triggers for man-db (2.7.0.2-5) ...
Setting up moonlight-embedded (2.2.0-1jessie1) ...
```

Am i missing something?

cgutman, 2016-04-08:
Did you ever do a "make install" before trying the package? Maybe it installed to a place that comes before the deb package install location in your path.

theFisher86, 2016-04-08:
cgutman, yes!  I did!  Do you know how to undo whatever I've done?

cgutman, 2016-04-08:
What do you get when you run: which moonlight

theFisher86, 2016-04-08:
`/usr/local/bin/moonlight`

cgutman, 2016-04-08:
Try deleting that file.

theFisher86, 2016-04-08:
IT WORKED!?! cgutman, irtimmer, dead you guys are the greatest human beings to ever walk the Earth!  Mother Theresa has nothing on you guys!!!!

Seriously, thanks so much.  I can never believe how much better quality support I receive on open-source stuff as opposed to things I pay for.