moonlight-stream/moonlight-embedded #362: Loss Stats: Transaction failed: 11

theFisher86, 2016-04-11:
Please provide the following info.

**GFE 2.11.2.67**
**Moonlight Embedded version: 2.2.0**
**Moonlight Embedded running on:** Raspberry Pi

**Output of Moonlight Embedded:** 
```
...
Starting input stream...done
Control Stream received disconnect event
Loss Stats: Transaction failed: 11
```

This is happening everytime I try to stream anything...

cgutman, 2016-04-11:
Can you upload the contents of your C:\ProgramData\NVIDIA Corporation\NvStream folder somewhere? Note that C:\ProgramData is a hidden folder.

theFisher86, 2016-04-11:
Here's the current log

http://pastebin.com/PVLcfT4K

cgutman, 2016-04-11:
#9(E)[2016-04-11 20:51:19,540]{000001F8} Failed to create FBC. Another instance exists and concurrent capture is not supported.

Do you have ShadowPlay turned on or some other software that would use hardware encoder and display capture?

theFisher86, 2016-04-12:
The newest version of GFE doesn't have ShadowPlay however this got me to thinking and I uninstalled KinoConsole and it fixed the problem.  KinoConsole is a program that also streams your desktop.  For anyone else who comes across this issue make sure you uninstall or disable any program that allows remote desktops or streaming.  I can confirm TeamViewer does not cause any issues.  Scan your Task Manager and processes and Services to see things that are currently running.

closed 2016-04-12 by irtimmer

herobank110, 2017-08-02:
What worked for me was turning on privacy control in the share option of GeForce Experience as this is constantly recording your desktop.

acid8x, 2017-08-13:
Same problem here, disconnecting itsel after +/- 1min. The only way I found to fix that was to set a lower bitrate. Default was 10000kbps, I set it to 5000