moonlight-stream/moonlight-embedded #363: No control with fallout 4

RealZack, 2016-04-12:
Don't know if the probleme come from mooblight or fallout.

i run it via Steam, the game start and i get the menu, but i don't have the mouse cursor and the keyboard doesn't work nether.

Strange fact ² key open the console and i can write letter and the cursor appear, but when I close the console nothing work.

Is it possible that the game is waiting for a controller instead of a keyboard/mouse?

Please provide the following info.

**_NVidia Geforce Experience version:_ *2.10*
**Moonlight Embedded version:*2.1.4*
**Moonlight Embedded running on:*Raspberry Pi 2 B*

**Output of Moonlight Embedded:**

**What is the expected result?*Controle the game like the others*

**What happens instead of that?*Nothing respond i. A normal use*

bertbert72, 2016-05-20:
Same problem here.  I've also noticed that I lose control on the host PC as well and have to alt-F4 it to get out.  That would maybe imply the problem is at the Gamestream side.

bertbert72, 2016-05-20:
Found the solution here:

https://forums.geforce.com/default/topic/894906/shield-android-tv/fallout-4-gamestream-/2/

Before running through Gamestream, launch Fallout 4 and in Settings > Gameplay and turn the controller off.

closed 2016-05-20 by irtimmer