moonlight-stream/moonlight-embedded #364: Not running fullscreen

MichaelRupertDev, 2016-04-13:
When running the newest version on moonlight-embedded on RPI3 running the newest version of RetroPie, moonlight doesn't capture all of the input or take the whole screen. When I press the keys on the keyboard, i see the input typed as if I were typing in the bash shell. The top line of the screen also displays a line of text.

Sardtok, 2016-06-28:
Yeah, I can't get it to work anymore. The solution I posted in #276: trying different combinations of host and client resolutions doesn't seem to help.

birefringence, 2017-02-06:
I see two behaviours:

1. When both systems (the PC and the RPI3) are connected to the same monitor, everything works fine.
2. When the RPI3 is connected to my projector, moonlight does not fill up the screen and the console is visible at the top and the bottom.

The only difference I can see is that the projector has a different aspect ratio (16:10) than my PC monitor (16:9). My hypothesis is that the streamed video always has the aspect ration of the PC screen (although the small sample size doesn't allow to make any kind of conclusion).

Crono141, 2017-02-09:
birefringence seems to be correct.  My host PC is connected to a 16x9 monitor, but my client is connected to a 16x10 and I have the same issue.  Changing the requested streaming resolution to a 16:10 aspect ratio does not fix the issue either.  Is this a problem with moonlight or with the nvidia streaming protocol in general?  Does anyone know a way to blank the bash shell running in the background?  Its a bit distracting.

Sardtok, 2017-02-13:
Blanking the terminal is a well-known fix for video playback when the video doesn't completely fill the screen:

`sudo sh -c "TERM=linux setterm -foreground black -clear all >/dev/tty0"`

To get text back:

`sudo sh -c "TERM=linux setterm -foreground white -clear all >/dev/tty0"`

This also clears all the text on the screen (i.e. scrolls so the terminal input is on the top line), as well as change the color of the text on the main screen. You can place those lines before and after the moonlight command in a launch script, or if you use SSH to log into the Pi you can just run them from the SSH session, as they will not affect your SSH session's terminal emulator.

tbergeron, 2017-04-13:
Any updates about this? I'm under the impression that it happened to me after a nvidia update. :-/

irtimmer, 2017-06-11:
The new version should automatically clear the screen and request to be shown fullscreen.

closed 2017-06-11 by irtimmer

cTatu, 2020-01-21:
It's possible this issue should be reopen. Only few of my games are streamed to full TV resolution regardless of the resolution of the game.

nVidia GeForce Experience Version: 3.20.2.34
Moonlight Embedded Version: 2.4.10

Resolution of main PC (nvidia) screen: 1280x1024 (4:3)
Resolution of TV: 1920x1080

Some games on fulscreen mode are streamed as 4:3 images to the TV and doesn't take the whole screen leaving 2 black spaces at the sides. Also some other games are streamed perfectly to the full resolution of the TV even on window mode which is crazy.

From the laptop which has the same resolution as TV (full hd) and using the Linux client of moonlight-qt those problematic games are streamed flawlessly. That is because the game image it's stretched horinzotally to make it fit the screen even the image has still a 4:3 proportion. But at least theres neves black spaces at the sides.

So the solution will be something like what moonlight-qt does, stretching the image. By the way I've tried all the tips given on all the related issues and all the combination of resolutions and aspect ratios of nvidia control panel and moonlight resolutions command (-720 -1080 and custom) and none of those make any difference. From all the tests I did, I conclude that the problem are the games themselves.