moonlight-stream/moonlight-embedded #365: No "real" 5.1 surround sound (HDMI)

alfalfa1978, 2016-04-15:
**_NVidia Geforce Experience version: 2.11.2.55
**Moonlight Embedded version: 2.2.0
**Moonlight Embedded running on: RPi 3 via RetroPie

**Output of Moonlight Embedded:**
Streaming works fine with no error messages.

**What is the expected result?**
working digital 5.1 surround sound

**What happens instead of that?**
Using the command line option -surround it does play sound, but my receiver only displays "pcm" and does not switch to digital 5.1.
I can manually switch to Dolby PL II but that's no true 5.1.
The -surround option fixes the screeching sound in Assassins Creed Syndicate and Dead or Alive 5 (intros) and it also gives me sound in Street Fighter V (without -surround there is no sound at all).
But in the case of SF V it misses the audio dialogues (intros) which are played through the center speaker on my PC.
Mortal Kombat X has the Dolby Digital Logo on title screen, but all of my games are only producing 2 channel PCM when streaming with Moonlight. Directly on my PC 5.1 audio is working great in all supported games.

The changelog says that OMX audio is added for surround sound, but I don't know how to enable OMX. Or is it just enabled by using the -surround option?

Do I have to change the client settings in Steam? I changed it already to "auto (5.1)" but I only get 2 CH PCM sound.

Unknownforce351, 2016-04-15:
OMX is used by default on a Pi, regardless of -surround or not.  You could force it further by putting -audio hdmi, but that option is used without specifying it, so it shouldn't matter.  The audio is output as Multi-Channel PCM, not as Dolby sound.  Dolby is an encoding that's not implemented in this application, so you won't get Dolby sound output.  It would be possible for OMX to convert the audio to Dolby, but it's not needed, and it would only add latency, causing audio lag.

How do you have your pi connected and TV and receiver?

alfalfa1978, 2016-04-16:
I have connected the Pi over HDMI to my Sony Bravia and the TV is connected with the receiver by an optical cable. 
With Kodi I get 5.1 sound, if it's supported by the source.

Unknownforce351, 2016-04-18:
Looks like surround is broken using newer gen GFE's.
Try uninstalling 2.11 and installing 2.9.  ( http://www.filehorse.com/download-nvidia-geforce-experience/old-versions/ )
I believe the second post here is what I used to prevent it from auto updating itself too: https://forums.geforce.com/default/topic/862779/how-do-i-disable-geforce-experience-auto-update-/

closed 2017-03-18 by irtimmer