moonlight-stream/moonlight-embedded #366: 50 FPS option possible? (enhancement)

alfalfa1978, 2016-04-15:
Is it possible to add an option to stream with 50 FPS?
50 FPS / Hz is supported by most displays and there are some games that can not hold the 60 FPS all the time on my PC with GTX 970 but never run below 50 FPS. So I use in some hardware demanding games like GTA 5 or AC Syndicate the 50 Hz VSync mode and it works just fine.
It would be nice if there would be an option to stream with 50 FPS instead of only 30 or 60.
It would also be a good compromise in relation to the needed bitrate and demanded GPU power of the Pi.

Unknownforce351, 2016-04-15:
Might sound like a stupid question here... because it might reject the setting... but wouldn't -fps 50 option do this?

alfalfa1978, 2016-04-16:
No, it does not. Only 30 and 60 work.

lexa2, 2016-04-16:
I agree with you that having 50fps would be a good thing but IMO moonlight-embedded issue tracker is a wrong place to ask for it. Moonlight is a receiver side, it requests GFE to send it the content with a specific encoding parameters and then processes the incoming stream. It can't request GFE to send a stream that is not supported by original Shield. I.e. it could be possible but chances are it won't work.

Correct place to ask for this feature is a nVIDIA's GFE support forum. As soon as nVIDIA would make 50fps available for original Shield - most probably it would take next to none efforts to support it on moonlight side.

alfalfa1978, 2016-04-17:
Thanks for your reply, lexa2.
Now I understand the background.

closed 2017-03-18 by irtimmer