moonlight-stream/moonlight-embedded #367: RTSP: Failed to connect to UDP port 48010

Septiron, 2016-04-16:
Hi there, 

since the GFE Update to 2.11 i got the RTSP error 11. After the moonlight-embedded update to 2.2.0  the error changed to another RTSP error:
`RTSP: Failed to connect to UDP port 48010
failed: -1`
So the fix of https://github.com/irtimmer/moonlight-embedded/issues/340 just changed the problem to another.

What i have done so far:
- Checked UPD port 48010 in my windows firewall. The port is by default open for nvstreamer.exe and is named on the firewall "SHIELD Streaming NvStreamer UDP Exception". 
- Check if the nvidia services are running
- Try uninstalling the driver using DDU in safe mode. And install the latest "normal" driver, choose "advanced" and check "clean install".
- Setting up a new Raspbian Jessie with moonlight-embedded.
- Disabled the firewall. The error changed to some 404 error.

Information about my stuff:
- Windows 10 Pro 64-bit (wired connected)
- NVidia Geforce Experience version: 2.11.2.66 (also tested it with 2.11.2.55)
- Moonlight Embedded version: 2.2.0
- Graphic Card: Nvidia GeForce GTX 760 (driver: 364.72)
- Raspberry Pi with Rasbpian Jessie (wired connected)
- The Streaming PC and Raspberry are on the same network.

Has anybody an idea?

Greetings Alex

cgutman, 2016-04-17:
Can you capture a Wireshark pcap file on your PC while you're trying to stream from your Pi and post a link to it here? The fact that the error changes if you disable your firewall suggests that the firewall is somehow involved here.

amylizzle, 2016-04-17:
I had this issue, and solved it by updating to 2.11.2.66. disabling Windows Firewall, and rebooting the host PC.

Septiron, 2016-04-17:
Thank you Amy, i just tested it, but it didn't solved my problem. The problem is the same (RTSP: Failed to connect to UDP port 48010). 

@cgutman I will take in the next days a pcap file from my streaming PC. The strange thing is, that my Nvidia Shield Tablet work with the streaming PC. So i didn't think that the firewall is blocking the UDP ports for streaming. FYI: Forget the 404 errorcode. I can't get it back or retrace it, when i disable the firewall. There is just the "RTSP: Failed to connect to UDP port 48010 failed: -1" error left.

Grettings Alex

Septiron, 2016-04-18:
Hi, here is the pcap file:
http://workupload.com/file/yJnCtMQ

cgutman, 2016-04-18:
I think I know what might be happening.

@Septiron Can you try Moonlight Android on your Shield Tablet and see if that works?

Septiron, 2016-04-19:
Hi, i tested it on my OnePlus X and Nvidia Shield and it works. 
So what is the different between you´re moonlight-android project and the embedded? :D

cgutman, 2016-04-19:
I think the bug is that nvstreamer.exe is not starting up in time for the connect packet from Moonlight. Since @irtimmer is using the stock version of libenet, it doesn't have my fix to make retransmissions work without multiple calls to enet_host_service(). This means only one connect packet is sent and if it is dropped, the connection will never succeed.

I've committed a patch to moonlight-common-c to fix this: https://github.com/moonlight-stream/moonlight-common-c/commit/d5e0950ec6fc3a803bf8918be72cff7463299ba0

Septiron, 2016-04-19:
Cool, i´m looking forward for the patch. Many thanks Cameron!

ronlaws86, 2016-05-06:
Is there a way around this problem? currently can't play games because of this error..

typinghands, 2016-05-08:
Any news on this patch being available? I am unable to stream on the raspberry, because of this error (streaming works only using android at the moment).

CanOfColliders, 2016-05-13:
Just noticed, if i stream to moonlight on android (on my phone) for just a moment, the RTSP 48010 error disappears on moonlight-embedded for me. Then i can then start streaming just fine until the next reboot of my pc. No actual solution, but maybe a helpful quick fix for some people.

ronlaws86, 2016-05-13:
My rig has 2 Network cards and i found that i had to disable one of them for it to work, not ideal but i found the RTSP error so far has gone. Could this be a binding or hostname resolve issue as well
?

typinghands, 2016-05-15:
@CanOfColliders I'm not sure if I'm doing the workaround correctly since it still gives me that error when I tried. What I did was: connected to android's app and streamed for a minute a Steam's game. Then I closed the app (and phone's wifi), and tried streaming the same game on Raspberry. Sadly it still froze at the RTPS handshake, did I do it incorrectly?

CanOfColliders, 2016-05-15:
@typinghands Hm, odd. That's what seems to work for me. Installed moonlight from the play store (the non-root version), connected to my pc, started the stream, closed the stream. Then streaming to moonlight-embedded (on a pi 3) seems to work without the RTSP error for me, until i do a reboot of the pc. I'm using the current GFE (2.11.3.5) on Win10.

typinghands, 2016-05-17:
I have done more experiments and noticed that I was mostly unable to stream if Steam was involved, and I was trying to launch a Steam game. The error didn't appear with a game like Tera instead, and I was able to stream it. A bit odd, especially since from Android's Moonlight I'm able to stream also Steam games. @irtimmer Any news on this bug getting fixed?

irtimmer, 2016-05-17:
Fix has been merged in version 2.2.1

closed 2016-05-17 by irtimmer

blast1987, 2016-10-12:
Hello, i am getting same error on GFE 3.0.7.34 and moonlight 2.2.2 
i have disabled windows firewall but still getting the error

Haapavuo, 2016-10-15:
Same here. Worked fine last week but now I'm getting this error.

sirus887, 2016-10-22:
Start getting this problem today. Thought it might have been due to having to connections on my host.

I on the most recent version of moonlight and GFE running from the most recent version of RetroPie via a wired(poweline) connection.

The work-around for me was to open up Steam, GFE, and UPlay(god damn ubisoft) on the host then run my moonlight script on pi. This seems to successfully run fine.

en00x, 2016-11-03:
I get the error too, but only when i choose a custom resolution. I want to run it in 1600x900, which my monitor is definitly able to show, and get the error. With default 720p there are no problems.

GFE 3.0.7.34 and moonlight 2.2.2

sirus887, 2016-11-03:
Last few days I have been getting the problem despite my workaround stated earlier. Another workaround someone had posted was too lock your computer and unlock it. Weird as it sounds, it works for me.

en00x, 2016-11-03:
Sadly no workaround solves this. But it is very mysterious that it works in 720p. I tried to reproduce it on android or PC, but in none of the apps for these platform i had the option to change the resolution to a custom one.

funk44, 2016-11-04:
Was working without problem for me until this evening when I got this issue?!?

Windows, Moonlight and GFE have not been updated in the past few weeks

sirus887, 2016-11-04:
@funk44   Have you tried the work-arounds I mentioned earlier? The lock unlock windows seems to work for me. I am on Win7, Latest GFE and Moonlight, running for Pi2.  I am running 1080 and 60fps too.

tofuSCHNITZEL, 2016-11-13:
I had the same problem. It helped to disable the second network adapter on my Win 10 x64 PC so the PC is now reachable under only one IP Address.

adilanchian, 2017-03-21:
Still having the same issue here =/. Running GFE 3.4.0.70. I am using the Java app on MacOS while my windows ten machine is the stream machine. I am only trying local network. I am able to connect and launch a game no problem, but it then closes with the RTSP handshake error with port 48010. Any help on this would be awesome!

MarcusVoelker, 2017-07-29:
Same here. Running moonlight 2.4 on my RPi and GFE 3.8.0.89 on my Win 8 PC.

Disabled firewall on PC, made sure to only have one wifi connection on the Pi, but I get that error when I try to connect, and then the whole Raspberry Pi locks (or at least becomes inresponsive to keyboard input), necessitating a hard reboot

MarcusVoelker, 2017-07-30:
Fixed by updating the drivers on PC and rebooting

paishin, 2017-07-30:
WiFi Adapter drivers?

MarcusVoelker, 2017-07-30:
Nvidia

Steuv1871, 2018-02-03:
Hi !
I've got the same issue and didn't manage to make this fraking UDP 48010 port to work.
But I've try to down grade Geforce Experience to a previous install (3.8.0.89, before they start using this port) and it works !
I've keept the install, I can share if you can get your hands on it, send me a private message.
Regards.