moonlight-stream/moonlight-embedded #368: Loss Stats: Transaction failed: 11

Kheiryk, 2016-04-25:
NVidia Geforce Experience version: 2.10.2.40
Moonlight Embedded version: 2.1.4
Moonlight Embedded running on: RPI2

Hello
When i start a game, i've this
[image]

it's the same issuie ?
https://github.com/irtimmer/moonlight-embedded/issues/299

If yes, without reinstall an old driver, it's possible to resolve it please ?
thanks

Kheiryk, 2016-05-01:
Hi all, anyone have resolved the same problem please ?

arv187, 2016-05-14:
Same issue:
```
pi@raspberrypi ~ $ sudo moonlight stream -app fifa -720 -30fps 192.168.1.103
Moonlight Embedded 2.2.0 (EMBEDDED;PI)
Connect to 192.168.1.103...
NVIDIA GeForce GTX 760, GFE 2.11.3.5 (protocol version 7)
Stream 1280 x 720, 30 fps, 5000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Discarding RTP packet queued for too long
Network dropped beginning of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Unmapped button: 306
Unmapped button: 306
Discarding RTP packet after queue overgrowth
Network dropped middle of a frame
Waiting for IDR frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped beginning of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped beginning of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped middle of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Unmapped button: 306
Unmapped button: 306
Discarding RTP packet queued for too long
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Discarding RTP packet queued for too long
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Unmapped button: 306
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Terminating RTSP handshake...done
Cleaning up platform...done
```

Kheiryk, 2016-05-24:
hi
no solutions ?

Kheiryk, 2016-05-26:
gfe : 2.11.3.5
moonlight : 2.2.0

not work always :/
and 

-- [ ERROR ] ------------------------------------------------------
USBController::USBController(): libusb_open() failed: LIBUSB_ERROR_ACCESS
$ sudo killall xboxdrv
$ xboxdrv

-- [ ERROR ] ------------------------------------------------------
USBController::USBController(): libusb_open() failed: LIBUSB_ERROR_ACCESS

$ sudo rmmod xpad
rmmod: ERROR: Module xpad is not currently loaded

Kheiryk, 2016-06-12:
Hi
I've make a new install (Raspbian, xboxdrv, moonlight, ...), but same problem.
do you have a solution please ?

thanks :'(

7hunderbug, 2016-06-13:
You do the pairing first with:

`moonlight pair 192.168.0.7` (press Enter key)

then, when prompted, put the 4 digit pin-code that moonlight generates on the Raspberry Pi console into the pin-code box in Geforce Experience on your PC...

...then you run moonlight like this:

`moonlight -mapping /home/pi/xbox.map -input /dev/input/event2 -1080 -app "Trine 3" stream 192.168.0.7`

Make sure you use the `-input /dev/input/event2` option, assuming your xboxdrv announces that it is set up to use /dev/input/event2, otherwise choose whichever /dev/input/event# it is using. If you don't use this option, moonlight might not detect your controller properly. Also, you don't have to have the `-app "Trine 3"` option; you can leave this option out and Steam will run automatically.

Kheiryk, 2016-06-13:
Hi and thanks

when i bash this command, i've always the transaction failed error, and the pad can't to be used.
but i see an other problem, i think, it's in relation.

- by putty, i make a **sudo xboxdrv,** and i push the buttons on the pas wired to the rpi2, no problem.
- but if i make **sudo xboxdrv,** directly on the terminal of raspbian, and i push the buttons, **no reactions ...**

7hunderbug, 2016-06-15:
You will always get the 'libusb_error_busy' error when running xboxdrv unless you remove the xpad kernel module first (or put xpad in a modules blacklist file) because they are two different drivers for the same Xbox controller and they conflict with each other if you try to use them at the same time.

Use EITHER xpad OR xboxdrv. I have found xboxdrv to function better than xpad.

If you want to use xboxdrv then ensure that xpad is unloaded from the kernel first by doing `sudo modprobe -r xpad` from the command line or, for a more permanent solution, by adding `blacklist xpad` to `/etc/modprobe.d/xpad-blacklist.conf` (create this file if it doesn't exist) then running `depmod -a` then reboot.

If you have chosen xboxdrv to drive your Xbox controller, you need to run xboxdrv either from a boot script like /etc/rc.local or from the console using sudo like this `sudo xboxdrv --silent &` before running moonlight.
Once xboxdrv is running, you can check which /dev/input/event# handler it is using by running `cat /proc/bus/input/devices` and looking for the `H: Handlers=` line corresponding to your `N: Name="Microsoft Xbox Controller` input device or by running `sudo evtest` if you have evtest installed. Once you know which /dev/input/event# handler your Xbox controller (xboxdrv) is using, you can then pass this as an argument to the moonlight-embedded `-input` switch.

Kheiryk, 2016-06-15:
Hi and thanks for your answer

i've reinstall a new Raspbian distrib, but i've just install xboxdrv after the updates, when i execute sudo xboxdrv, i've already libusb-error-busy
and there is no the XPAD driver, i've not install it

but i continue your tuto, thanks

7hunderbug, 2016-06-15:
The xpad kernel module/driver is part of the kernel and is installed by 
default - double check the loaded modules with `lsmod | grep xpad` - 
this will show xpad listed if it is loaded in the kernel. The xpad 
driver will load a soon as you plug in your Xbox controller (either 
wired or wireless dongle) via USB, it doesn't require to be installed.

Kheiryk, 2016-06-15:
Hi and thanks
this part doesn't work  
`by adding blacklist xpad to /etc/modprobe.d/xpad-blacklist.conf (create this file if it doesn't exist) then running sudo depmod -a then reboot.`

I reboot and i execute 
`lsmod | grep xpad`

but xpad is always start
**xpad                   11952  0**

i execute
sudo modprobe -r xpad
and sudo modprobe -r xpad, it's work, xpad is close ...

**I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="Xbox Gamepad (userspace driver)"
P: Phys=
S: Sysfs=/devices/virtual/input/input3
U: Uniq=
H: Handlers=event2 js0**

evtest
/dev/input/event0:        Mini Keyboard
/dev/input/event1:        Mini Keyboard
/dev/input/event2:      Xbox Gamepad (userspace driver)

and the pad works !!!
just if i can bash an autostart for blacklist xpad, it's better :)

thank you !!

7hunderbug, 2016-06-15:
I'm glad you got your controller working. To check you've got the xpad module blacklisting correct, can you post the output of:
`cat /etc/modprobe.d/xpad-blacklist.conf`
please?
It should only have one line like this:
`blacklist xpad`

Your rc.local line to run xboxdrv automatically looks correct.

Do any games work or do you always get disconnected? If you always get disconnected, try running moonlight without the `-1080` switch and without the `-app "Trine 3"` switch so that it defaults to 720p and Steam and report back as to whether Steam works as expected.

The error relating to the "disconnect event" and the "Transaction failed" could be related to the game you are trying to run but are most likely related to a problem with either your network, your PC configuration, your Windows install or a problem with the NVidia drivers and GFE. It is almost definitely *not* a problem with moonlight-embedded or your Raspberry Pi, especially considering you have stated that you've done a clean Raspbian install with the most up to date moonlight-embedded. I would suggest that the most likely culprits are the Nvidia drivers and GFE install as this is the area where I've had most problems and my issues have always been solved by doing a completely clean install of the NVidia components (use Expert/Advanced option in Nvidia installer then tick box for clean install).

Unfortunately, this isn't the place to be troubleshooting PC, Windows or network issues.... however, you could try any of the following:

1. disable IPV6 on your PC network interface
2. disable IPV6 on your router (if it is supported and enabled)
3. ensure only one of your PC network interfaces is connected (e.g. disable wireless interface on your PC so that only the wired NIC is being used)
4. try moonlight on your Android phone if you have one
5. disable IPV6 on your Raspberry Pi. see https://www.leowkahman.com/2016/03/19/disable-ipv6-raspberry-raspbian/
6. disable your Windows firewall and/or antivirus temporarily to test streaming without those
7. remove/disable any remote desktop software that might be running on your Windows PC
8. check moonlight-embedded github closed issues for other suggested solutions: https://github.com/irtimmer/moonlight-embedded/issues?q=is%3Aissue+is%3Aclosed

Kheiryk, 2016-06-15:
Hi

```
~ $ cat /etc/modprobe.d/xpad-blacklist.conf
blacklist xpad
```
without -1080, doesn't work too

Thanks

7hunderbug, 2016-06-15:
Blacklisting should work so something is wrong with your Raspbian install. Are you running anything else on the clean Raspbian install that you haven't mentioned, like Retropie or Kodi or something?

I've just tried blacklisting xpad on my Raspberry Pi and it works perfectly. Your problem might be that there are invisible characters in your blacklist file - first delete the file: `sudo rm -rf /etc/modprobe.d/xpad-blacklist.conf`, then create it again: `sudo nano /etc/modprobe.d/xpad-blacklist.conf`. Do not copy and paste `blacklist xpad` into the blacklist file, type that line out manually.

What do you mean by STEAMLINK? Do you mean that you have a Steamlink device connected to your TV and SF5 does not work with that device either?
Does Steam run via moonlight? Can you navigate through Steam? Can you visit store pages and watch the promotional videos within Steam? Do you get disconnected from Steam?
What do you mean "without -1080, doesn't work"? Do you mean nothing works at all or do you mean you get disconnected after a short time?

Please try to be more explicit and specific with your information!

7hunderbug, 2016-06-15:
OK, `-1080` has got nothing to do with your pad not working, so either your moonlight command line is incorrect or because you are not starting your game directly (you have omitted `-app "Trine 3") then Steam is not detecting your pad.

If Steam starts up in big picture mode you can check which controllers are detected by going to the relevant settings in the Steam settings window.

Sometimes controller problems can occur, where Steam can't detect a controller. if you have another controller connected to your PC - please ensure you have *no* other controllers connected to your PC, then reboot your PC and try again.

7hunderbug, 2016-06-15:
Also, you should create a new moonlight mapping file for your controller, especially considering that your current mapping file was probably created when the xpad driver was being used and if that is the case, your mapping file will be incorrect.

Kheiryk, 2016-06-15:
> Blacklisting should work so something is wrong with your Raspbian install. Are you running anything else on the clean Raspbian install that you haven't mentioned, like Retropie or Kodi or something?

no, just moonlight, xboxdrv et evtest

>  first delete the file: sudo rm -rf /etc/modprobe.d/xpad-blacklist.conf, then create it again: sudo nano /etc/modprobe.d/xpad-blacklist.conf. Do not copy and paste blacklist xpad into the blacklist file, type that line out manually

same problem, strange

> What do you mean by STEAMLINK? 
steam://open/bigpicture (a shortcut for steam)

> What do you mean "without -1080, doesn't work"? Do you mean nothing works at all or do you mean you get disconnected after a short time?
just the pad can't be used

> Does Steam run via moonlight? Can you navigate through Steam? Can you visit store pages and watch the promotional videos within Steam? Do you get disconnected from Steam?

by my link, yes

> If Steam starts up in big picture mode you can check which controllers are detected by going to the relevant settings in the Steam settings window.

i can use the pad in Steam menu

> Sometimes controller problems can occur, where Steam can't detect a controller. if you have another controller connected to your PC
yes i've an other controller :/ i can test without this

i check
thanks ;)

Kheiryk, 2016-06-15:
In steam (big picture) parameters, i see "controller Xinput"
i've unplug the controller in the host, but same problem

7hunderbug, 2016-06-15:
And the problem now is what? Your controller doesn't work at all? Did you make a new mapping file for moonlight, now that you're using xboxdrv?

Kheiryk, 2016-06-15:
> Your controller doesn't work at all?
jus on each games

i've try to use my xbox.map
for example the pad on Trine 3 work perfectly (with and without steam)

thanks

Kheiryk, 2016-06-16:
Hi
i've put blacklist xpad in a blacklist.conf and the raspi-blacklist.conf, but don't work, xpad is always start.
But now, i can play with 3 games / 4, with xpad start (without make rmmod xpad), strange ...

maybe make a bash for execute rmmod xpad in start of Raspbian ?

just for the thirth game, i must clic with my mouse before on the good menu of this game (walking dead)
but when i play to the game and i quit the game, i can navigate in the menu with my pad ...

my xbox.map
`
abs_deadzone = 0
abs_dpad_x = -1
abs_dpad_y = -1
abs_rx = 3
abs_ry = 4
abs_rz = -1
abs_x = 0
abs_y = 1
abs_z = -1
btn_dpad_down = 295
btn_dpad_left = 296
btn_dpad_right = 297
btn_dpad_up = 294
btn_east = 305
btn_mode = 316
btn_north = 308
btn_select = 314
btn_south = 304
btn_start = 315
btn_thumbl = 317
btn_thumbr = 318
btn_tl = 310
btn_tl2 = 312
btn_tr = 311
btn_tr2 = 313
btn_west = 307
reverse_dpad_x = false
reverse_dpad_y = false
reverse_rx = false
reverse_ry = true
reverse_x = false
reverse_y = true
`

SF5 don't work in menu and in game :'(
i make some tests

thanks

Stijn98s, 2016-10-22:
I have the same issue. with no controllers and i just want to start Steam. I already reinstalled GFE and clean installed the drivers, both on ethernet connection and disabled wireless internet driver and i have disabled firewall. I have removed TeamViewer and closed Razer Synapse. I have tried it on raspberry pi 3 and my Android Mobile.

Moonlight embedded: 2.2.2
GF Experience: 3.1.0.44
GF Game Ready Driver: 375.57

PC:
Windows 10 Home 64 bit
GeForce 840M
i5-4200U

link123a, 2017-02-15:
Hello!
I was having this same problem as well ( just got moonlight installed on raspbian)
by running [ moonlight stream "ip of host computer" ] I was connecting but getting disconnected.

I figured out it's because I was using remote desktop application to monitor my host computer ( Splashtop Personal ). As soon as I closed the remote desktop application, and ran moonlight again, every connected and it worked! 

I hope this helps!

closed 2017-03-18 by irtimmer