moonlight-stream/moonlight-embedded #371: Keyboard Issue in steam big picture mode/gaming

sheldon25, 2016-04-28:
Please provide the following info.

**_NVidia Geforce 2.11.2.66**
**Moonlight Embedded It is current version Just downloaded it yesterday. Sorry I am not at my pi and can not remember the numbering*
**Moonlight Embedded running on:** Raspberry Pi 3 Rasbian Jesse
**Output of Moonlight Embedded -720 -30fps:**

**What is the expected result?** Game to run with expected latency of wireless in home network

**What happens instead of that?**Steam big picture will open up, after scrolling through the library ,settings or a few minutes in a game it will seem to stick a certain input like direction left or down. I have tried this when I had a ps3 controller hooked up and mapped in addition to a wireless keyboard. Same result, tried this without a ps3 controller hooked up and just the wireless non bluetooth keyboard and i will get the same result. Has anyone encountered this? If there is any extra info you need let me know. I am not 100% versed in linux. Started with the pi 2 and openelec, I can get retropie to work flawlessly but am having issues with moonlight.

cricket007, 2016-05-01:
Sounds like #264

sheldon25, 2016-05-02:
Thanks, this resolved my issue!!

closed 2016-05-02 by irtimmer