moonlight-stream/moonlight-embedded #372: Not recognizing full input in-game, but works in Steam menus

cricket007, 2016-05-01:
I'm having a strange issue where I am able to use the controller (Sony PS3 Controller) perfectly fine to navigate within Big Picture Mode and the Steam menus while in-game (accessed via the PS button), but some games<sup>1</sup> (with partial controller support) are either completely not accepting the controller input or, in particular for one game<sup>2</sup>, certain types of menus act like they'll "take over" inputs. All other events in that game accept the correct inputs, so I'm assuming keyboard events are being expected, but I am unable to check that, as I don't have a keyboard attached to my Raspberry Pi and I'm not sure how attaching one would help since I have applied a mapping file for my controller. 

I currently have the controller connected by bluetooth (also tried wired, but prefer bluetooth) and have triple checked the controller mapping (plus, as I said, the inputs mostly work). It should also be noted that all inputs for these games work as expected when the same controller is directly wired to the PC and is not streaming. 

**_NVidia Geforce Experience version:_** 2.11.2.55
**Moonlight Embedded version:** 2.2.0
**Moonlight Embedded running on:** Raspberry Pi 3 - Running OSMC (Debian Jessie)

**Output of Moonlight Embedded:** 
The streaming runs perfectly fine. Using the Luna launcher for Kodi, but I am able to SSH and run moonlight as well. 

**What is the expected result?** 
All inputs for streaming and a wireless controller act the same as when directly wired. 

**What happens instead of that?**
Certain games<sup>2</sup> accept full input except in certain menus while other games<sup>1</sup> do not accept the controller input at all. 

<sup>1</sup> Final Fantasy XIII & Final Fantasy XIII-2
<sup>2</sup> Final Fantasy XIII-3

wilsonic, 2016-05-08:
Did you follow these [instructions?](https://github.com/moonlight-stream/moonlight-android/wiki/Setup-Guide#using-a-gamepad-connected-to-the-pc-instead-of-the-streaming-device)

cricket007, 2016-05-08:
@wilsonic I saw that, but I did not follow them because I need the controller to be connected to the streaming device. Reason being is I don't have a usb Bluetooth adapter; the raspberry pi 3 has an embedded one.

Unknownforce351, 2016-06-02:
Unplug any/all gamepad devices from the host.

cricket007, 2016-06-02:
@Unknownforce351 All I have plugged in are a keyboard, mouse, and external HDD. You suggest removing those?

Unknownforce351, 2016-06-02:
Well unplugging the mouse and keyboard might help, but it's not a solution, just a workaround.  Go into Devices and Printers control panel and and remove any/all game controllers listed there, regardless of whether or not they are plugged in, disabled, or bluetoothly paired but not active (such as Nintendo Wi-Motes or other bluetooth controllers that you've paired with the PC)

cricket007, 2016-06-26:
I'm not sure what I did, but everything works as expected now. :smile:  

I did see an update to `moonlight-embedded` during a `apt-get upgrade`, but believe it was a combination of using the `sixad` service (instead of bluez? ... not sure if those services are related or conflict in any way), or I re-wrote my mapping file manually instead of relying on `moonlight map` by getting button codes from installing `evtest`.

cricket007, 2016-06-26:
For reference, here is the mapping file that I now have. 

Controller: Sony Dualshock 3, Bluetooth, QtSixA

```ini
abs_x = 0
abs_y = 1
abs_z = -1
reverse_x = false
reverse_y = true
abs_rx = 2
abs_ry = 3
abs_rz = -1
reverse_rx = false
reverse_ry = true
abs_deadzone = 0
abs_dpad_x = -1
abs_dpad_y = -1
reverse_dpad_x = false
reverse_dpad_y = false
btn_north = 300
btn_east = 301
btn_south = 302
btn_west = 303
btn_select = 288
btn_start = 291
btn_mode = 304
btn_thumbl = 289
btn_thumbr = 290
btn_tl = 298
btn_tr = 299
btn_tl2 = 296
btn_tr2 = 297
btn_dpad_up = 292
btn_dpad_right = 293
btn_dpad_down = 294
btn_dpad_left = 295
```

closed 2016-06-26 by cricket007