moonlight-stream/moonlight-embedded #374: Xbox 360 wireless controller not working correctly

klen-ghost, 2016-05-16:
Please provide the following info.
Is a retropie installation
**_NVidia Geforce Experience version: 2.10.2.40
**Moonlight Embedded version: 2.1.4
**Moonlight Embedded running on:** _Raspbery pi 2

**Output of Moonlight Embedded:** 
Don't know (how do i check this?).
Inside retropie xbox 360 controller works as should. Just steam bpm only left and right inputs are taken. Cant navigate to games to test.

**What is the expected result?**
Controller working as it should in stam BPM and games
**What happens instead of that?**
Only left and right d-pad navigation works in steam BPM

RikJm, 2016-05-29:
Can you paste how you're launching moonlight? I believe this is related to the input flag being used before stream? or something like that...

Try launching it like this:
#!/bin/bash
sudo ./moonlight stream -720 -30fps 192.168.0.3 -mapping xbox360.conf

klen-ghost, 2016-05-30:
Trying to launch it via the luna launcher addon.
There seems to be no xbox.map file on the sd card.
Can't make one via map xbox.map command ( no controller input is registerd
when i press any of the buttons, wireless controller is connected via a 3th
party xbox 360 receiver).
Will try to make a mapping file via "sudo nano xbox.map"
However on another sd card which contains recalbox i can acces steam and
everything works (so everything else works)
Thanks for trying to help!!

RikJm, 2016-05-30:
oh sorry, I assumed you'd have a .map or .conf file. 
You could create one with moonlight by typing `sudo moonlight map nameofthefile.conf`
I don't know how you're launching moonlight but I remember from a while back that if you use -input before stream things can go wrong with the controller, make sure you don't have that.

klen-ghost, 2016-05-30:
i tried creating a map file but there is no input from the controller taken.
i don't know how they set up that moonlight launcher shortcut in kodi
addon. ( so there for i can't tell you if the input line is placed wrong).

tried starting via SSH via

#!/bin/bash
sudo ./moonlight stream -720 -30fps 192.168.2.4 -mapping xbox.map

getting the following error :  sudo: ./moonlight: command not found

my dir looks like this (fyi) :

osmc@osmc:~$ dir

Movies                                 libevdev2_1.4.6+dfsg-1_armhf.deb

Music                                  master.zip

Pictures                               moonlight-embedded-2.1.4

RetroPie                               moonlight-embedded-2.1.4.tar.xz

RetroPie-Setup                         moonlight-osmc

TV\ Shows                              openvpncert.ovpn

cmake-3.3.1                            rbp-moonlight-osmc_0.3.3_all.deb

cmake-3.3.1.tar.gz                     script.moonlight-osmc-0.3.3.alpha.zip

cmake-3.3.1.tar.gz.1                   script.openvpn-2.1.0.zip

evtest-1.25.tar.bz2                    tar

install-retrosmc.sh                    xbox.conf

libevdev-dev_1.4.6+dfsg-1_armhf.deb    xbox.map

libevdev-tools_1.4.6+dfsg-1_armhf.deb  xbox.map.save

hope this will give an hint of what is possibly wrong on my side (kind of
new to all of this stuff)

thanks for the great support!

RikJm, 2016-05-30:
oh sorry, my bad again, the ./ in front of the moonlight will only work if you have moonlight on that directory, so you could either cd to the one moonlight lives or if you installed kodi using the apt-get then you could just use `moonlight` without the ./ in front of it.

Try using `moonlight pair nameoffile.conf` via SSH and if that works and you're able to save a .conf file then try launching the stream also via SSH

klen-ghost, 2016-05-30:
can you tell me how to install moonlight into the directory ./moonlight dir
? (maybe it will work then and all will be easyer for both (since that
would be more standard then my install  ;) ).
sadly i can't errase my sd at this time since i don't know how to install a
vpn client and files on my own ;)

RikJm, 2016-05-30:
I'll try to explain as much as I can, I'm sadly not an expert on this.

./ is used when you want to launch something that's on the current directory you're in. For example I see you have a moonlight-embedded-2.1.4 directory on your file system. So, if you were to cd (change directory) to your moonlight.....blah directory once there you could ./moonlight and it would work.

For what I see you must have downloaded the release and unpacked it yourself rather than using apt-get, now moonlight is able through repository, if you want you can follow this tutorial on how to install it throug apt-get (way easier imo) [https://www.raspberrypi.org/magpi/stream-steam-games-to-your-raspberry-pi-tutorial/](url)

Another way of doing it would be to create a .sh file that changes directory and then launches moonlight, something like this:
`#!/bin/bash
cd /home/osmc/moonlight-embedded-2.1.4/ && ./moonlight stream 192.168.2.4 -mapping /home/osmc/xbox.map`

take a look how I've specified teh /home/osmc/xbox.map folder path too, if you don't it will assume it's on your moonlight directory, it's also possible but the xbox.map would need to be copied there.

I hope some of this can help you a bit.

klen-ghost, 2016-05-30:
tried it via the link suplied.
can't pair ( device is already paired) , i don't know how to resolve my
issue (maybe i need to take my old sd card and try starting from scratch
with a clean osmc install with only exodus and vpn installed, and start
with moonlight via the apt-get method.

starting to understand how linux- ssh works bit by bit (to much used to
windows i gues)
sure your input does help me to understand how it works and all :)

RikJm, 2016-05-30:
I'm pretty sure moonlight has an unpair option `moonlight unpair 192.168.2.4` however if that doesn't work for some reason, you can unpair via the geforce experience settings tab [https://s3-us-west-1.amazonaws.com/shacknews/assets/editorial/2014/12/Preference%20Gamestream.jpg](url) you can see it in this image.
Try with and without sudo too, today I was having the: Failed to connect to UDP port 48010 error and pairing and launching with sudo, helped with that. (I have no idea why and maybe it wasn't just that but a combination of reinstalls and reboots, I don't know enough to tell, I just try :P)

klen-ghost, 2016-05-30:
managed to unpair and pair again.
strange thing is when i enter command :
moonlight stream (ip adres)
keeps telling me that i need to pair first (just did that ofcourse)
guessing that a clean sd card with fresh osmc install and install a fresh
copy of moonlight would be the best option ( maybe it will be running
better and faster afterwards, since the spare card is an uhs 3 card instead
of a regular class 10)

RikJm, 2016-05-30:
Don't give up just yet, make sure when you pair you then call moonlight with the same user, so, if you did `sudo moonlight pair (ip)` you then need to `sudo moonlight stream (ip)` and if you did `moonlight pair (ip)` then you can use `moonlight stream (ip)` sudo is the superuser while calling moonlight without it will use your osmc user instead.

klen-ghost, 2016-05-30:
tried all  combo's
sudo pair  - sudo stream
sudo pair - stream
pair - sudo stream
pair - stream

keep getting "need to pair first message"
(paired both sudo and non sudo user, just 2 be sure).

getting stuck with this ( used to work but that was on a recalbox
insttallation, but i'm not able to install a vpn client on that card).

where are you from (which country?)

cgutman, 2016-05-30:
That pairing message indicates that you're not running the latest Moonlight version. Can you paste the output of "moonlight help"?

klen-ghost, 2016-05-30:
as you asked the moonlight help output :

--------

Moonlight Embedded 2.1.4 (EMBEDDED;PULSE;PI)
Usage: moonlight [action] (options) [host]
       moonlight [configfile]

 Actions

        map                     Create mapping file for gamepad
        pair                    Pair device with computer
        stream                  Stream computer to device
        list                    List available games and applications
        quit                    Quit the application or game being streamed
        help                    Show this help

 Global Options

        -config <config>        Load configuration file
        -save <config>          Save configuration file

 Streaming options

        -720                    Use 1280x720 resolution [default]
        -1080                   Use 1920x1080 resolution
        -width <width>          Horizontal resolution (default 1280)
        -height <height>        Vertical resolution (default 720)
        -30fps                  Use 30fps
        -60fps                  Use 60fps [default]
        -bitrate <bitrate>      Specify the bitrate in Kbps
        -packetsize <size>      Specify the maximum packetsize in bytes
        -remote                 Enable remote optimizations
        -app <app>              Name of app to stream
        -nosops                 Don't allow GFE to modify game settings
        -localaudio             Play audio locally
        -surround               Stream 5.1 surround sound (requires GFE 2.7)
        -keydir <directory>     Load encryption keys from directory

 I/O options

        -mapping <file>         Use <file> as gamepad mapping configuration
file
                       (use before -input)
        -input <device>         Use <device> as input. Can be used multiple
time
                      s
        -audio <device>         Use <device> as ALSA audio output device
(defaul
                        t sysdefault)
        -forcehw                Try to use video hardware acceleration

Use Ctrl+Alt+Shift+Q to exit streaming session

osmc@osmc:~/moonlight-embedded-2.1.4$
-----------

cgutman, 2016-05-30:
So you've got the package also installed via apt-get?

Try running 'which moonlight' then 'rm [path from previous command]'

klen-ghost, 2016-05-30:
/usr/local/bin/moonlight
is output from which moonlight
will try to rm

RikJm, 2016-05-30:
Now with cgutman here you're in good hands, I'm gonna keep an eye on this post see if I can learn something :)

klen-ghost, 2016-05-30:
thanks haha. i'm learning more and more (maybe i should pick up on my
programming skills  whahaha... only completed a html home study about 7
years ago)

cricket007, 2016-06-02:
In the Luna settings in Kodi, there is a setting for you to select a mapping file. This will get passed to the `-input` flag of the `moonlight` command. 

But the pairing problem may be related to the VPN (which you shouldn't need while using moonlight). When you use `moonlight pair <IP>` you must go over to the PC and enter the PIN. If the PIN dialog does not appear, you probably need to go into the Geforce Experience settings and remove your paired devices and restart the process from there.

klen-ghost, 2016-06-03:
The pairing problem had to do with gfe (unit was already registered to pc,
got that fixed)
Ps. The Vpn is not used for luna but streaming purpose.

Will try to get xbox 360 mapping file to work via the luna addon. Will post
results  after trying.

closed 2017-03-18 by irtimmer