moonlight-stream/moonlight-embedded #375: Can't uninstall MOONLIGHT, please help!

keem85, 2016-05-16:
Hi, I did an "sudo apt-get remove moonlight-embedded" and also purged the package.. Still it pairs, and still I can access the moonlight-menus by typing "moonlight" in SSH..

I think this is because way back, I installed a moonlight-embedded version 2.1.1 using cmake commands from here: http://jordanhotmann.com/guide-install-moonlight-on-raspbianretropie/

Problem is that I can't remove the program at all.. I need to remove it to be able to use the apt-get installer version instead.. When I install moonlight via apt-get, it will not pair, because when I do the pair, it chooses the old (cmake) one instead.. How can I fix this?

TLDR: I can install/remove moonlight as I please, but I can't pair it, because there is an old CMAKE version of moonlight somewhere on my system....

-------------------------------------------
Please provide the following info.

**_NVidia Geforce Experience version: 2.11.35**
**Moonlight Embedded version: Newest**
**Moonlight Embedded running on:**  Raspbian RP2

**Output of Moonlight Embedded:**

**What is the expected result?**

**What happens instead of that?**

cgutman, 2016-05-16:
Yeah, another user ran into this issue too.

Run the command: which moonlight

then: sudo rm <path from the above command>

keem85, 2016-05-16:
I am forever in your debt! :) Thanks a lot!

closed 2016-05-16 by irtimmer

t1gu1, 2022-08-28:
In the worst case scenario, if you install it as a snap package:
`snap remove --purge moonlight`