moonlight-stream/moonlight-embedded #376: Cannot update to 2.2.1

scapeg0at, 2016-05-20:
Please provide the following info.
after downloading the newest release via wget
`wget https://github.com/irtimmer/moonlight-embedded/releases/download/v2.2.1/moonlight-embedded-2.2.1.tar.xz`
and following the instructions on the wiki
`mkdir build
cd build/
cmake ../
make
make install`
When I run moonlight its still says I have version 2.2.0 install and when I run `moonlight stream` give me the error
`RTSP: Failed to connect to UDP port 48010
failed: -1`
as discussed in #367

typinghands, 2016-05-21:
I'm also having some difficulties, but mostly compiling it because of missing dependencies.

Is the new version 2.2.1 also available through the Raspian repository, because I didn't manage to find it (I added the repository in the sources.list). If not so, how long does it usually take to become available for download there? 
Sorry, I'm still pretty new with linux and raspberry.

alfalfa1978, 2016-05-21:
Got the same problem - I updated, but it still says 2.2.0. But it works!

after "make install" it says:

-- Install configuration: ""
-- Installing: /usr/local/lib/libmoonlight-pi.so
-- Set runtime path of "/usr/local/lib/libmoonlight-pi.so" to ""
-- Installing: /usr/local/bin/moonlight
-- Set runtime path of "/usr/local/bin/moonlight" to ""
-- Up-to-date: /usr/local/share/moonlight/mappings
-- Up-to-date: /usr/local/share/moonlight/mappings/default.conf
-- Up-to-date: /usr/local/share/moonlight/mappings/dualshock3.conf
-- Up-to-date: /usr/local/share/moonlight/mappings/xbox360.conf
-- Up-to-date: /usr/local/share/moonlight/mappings/dualshock4.conf
-- Up-to-date: /usr/local/share/moonlight/mappings/rumblepad2.conf
-- Up-to-date: /usr/local/etc/moonlight.conf
-- Installing: /usr/local/lib/libgamestream.so.2.2.0
-- Up-to-date: /usr/local/lib/libgamestream.so.0
-- Up-to-date: /usr/local/lib/libgamestream.so
-- Set runtime path of "/usr/local/lib/libgamestream.so.2.2.0" to ""
-- Installing: /usr/local/lib/libmoonlight-common.so.2.2.0
-- Up-to-date: /usr/local/lib/libmoonlight-common.so.0
-- Up-to-date: /usr/local/lib/libmoonlight-common.so
-- Installing: /usr/local/share/man/man1/moonlight.1

Always 2.2.0 - is it just a wrong version displayed, or what went wrong?

irtimmer, 2016-05-22:
Sorry, forgot to update the version number in CMakeLists.txt :P

scapeg0at, 2016-05-22:
How would that problem be solved on my personal machine? Should I just wait until you can fix it or is there any easy changes to the files I can make? (bear in mind I'm no expert in any of this)

Septiron, 2016-05-26:
I also can't get the update 2.2.1, so i´m still getting the RTSP error.

bloo-dye, 2016-06-02:
Streaming with a higher resolution than that of the monitor connected to the PC causes the RTSP error. I used my PC's monitor to test the raspi, leaving my PC with a 1024x768 monitor. That gave me the exact same error when streaming with default settings. Maybe thats case for you too.

typinghands, 2016-06-15:
Thank you for sharing @bloo-dye, after removing the -1080 entry from the stream command I can now successfully stream from pc! I think this might be a good solution to some people.

closed 2017-03-18 by irtimmer