moonlight-stream/moonlight-embedded #379: Stream works on monitor but not on TV.

damadmax, 2016-06-05:
**NVidia Geforce Experience version:** 2.11.3.5
**Moonlight Embedded version:** 2.2.0
**Moonlight Embedded running on:** Raspberry Pi 3

When the Pi is connected to my 2nd monitor right next to the PC everything is working as expected.
But when i move the Pi into the living room and connect it to my TV only sound is transmitted to the TV.
The connection in both cases is wired with a 1gbit network. so i don't think it's a bandwidth issue.
The TV is HD-ready only, but i used the -720 switch to be sure.

Unknownforce351, 2016-06-06:
You get video with everything else on the Pi?

With it connected to the TV, run these two commands and post the output from it:
```
tvservice -d edid.txt
edidparser edid.txt
```

Also, what model TV are you using?  How do you have the Pi connected to the TV?  Using any Receiver/Amp in between?

Unknownforce351, 2016-06-06:
Looks like it doesn't support 1080p from what I'm seeing.

I'm not sure which will likely work (if any) here, but try the following:
Set your computers Desktop Resolution to 720p.
Remove all non-essential command line parameters, just do moonlight stream -app XYZ, the default is 720p @ 60fps I believe, so you shouldn't have to specify anything extra.
Re-pair moonlight (probably won't help, but won't hurt to try either!)
Not sure how savvy you are with linux, but open /boot/config.txt (sudo nano /boot/config.txt)
and add the following to force the Pi to use 720p:
```
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=4
hdmi_drive=2
#These are (probably) not related, but probably helpful regardless
gpu_mem=256
dtparam=audio=on
hdmi_ignore_cec_init=1
```

This should force it to use CEA mode 1280x720p @ 60 Hz
The others set gpu memory available to 256MB, which is plenty from the Pi3's available resources, makes sure it's compatible with the latest kernel and if you ever plug in to a CEC enabled device, will stop it from switching inputs or turning on/off the cec enabled device when booting up. These likely will not help your problem, but won't hurt anything and help other potential problems in the future.

Let me know if any of that helps.

closed 2017-03-18 by irtimmer