moonlight-stream/moonlight-embedded #381: RTSP handshake failed to connect to UDP port 48010 failed -1

SlimSterling, 2016-06-13:
Using GFE version 2.11.3.5
Moonlight Embedded version 2.2.0
Running on a raspberry pi

Whenever i do moonlight stream <I.P> -1080 -30fps i get 

RTSP handshake failed to connect to UDP port 48010 failed -1

Using a wired connection if that changes anything.
And installed from the packages list on the official wiki

Crypticat, 2016-06-15:
I have the exact same issue. The default -720 option works fine. I even tried adjusting the bitrate and that didn't help but when set too high, I get the same problem on the default 720 resolution. Running the latest GFE but I don't think it is that.

Septiron, 2016-06-27:
I also reported the issue here: https://github.com/irtimmer/moonlight-embedded/issues/367#event-663414154
The solution of the issue is to update to Moonlight Embedded version 2.2.1. 
But as you can see in this issue https://github.com/irtimmer/moonlight-embedded/issues/376, the update to 2.2.1 didn't work. 
I guess the only solution is to wait for the update 2.2.1? :-(

closed 2016-06-27 by irtimmer