moonlight-stream/moonlight-embedded #382: E: Unable to locate package moonlight-embedded

Droid22, 2016-06-13:
I am simply trying to install moonlight. I updated my sources.list so that it looks like this.
`deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi
 Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspbian.org/raspbian/ jessie main contrib non-free rpi
deb http://dl.bintray.com/kusti8/chromium-rpi jessie main
deb http://archive.itimmer.nl/raspbian/moonlight jessie main
`

After I run the command `sudo apt-get install moonlight-embedded` 
I get the error `E: Unable to locate package moonlight-embedded`

irtimmer, 2016-06-13:
Have you run `sudo apt-get update`?

Droid22, 2016-06-13:
yes

Droid22, 2016-06-14:
I have tried reinstalling raspian. adding the source again then doing sudo apt-get update and still the same results

dkeppens, 2016-06-17:
Are you sure the repo you added hasn't got a typo ? It says itimmer. Shouldn't it be irtimmer ?

dkeppens, 2016-06-21:
Nevermind my previous comment. The repo syntax is correct, i just installed from, it myself to correct the problem i've been having with my compiled master version.
Can you perhaps post your exact /etc/apt/sources.list file again, this time properly formatted so it's more readable ? Here's mine. apt-get update and apt-get install moonlight-embedded worked fine for me

root@keppiePI:/home/dkeppens# cat /etc/apt/sources.list
deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi
#following line for kodi 16.1 package
#deb http://pipplware.pplware.pt/pipplware/dists/jessie/main/binary /
deb http://archive.itimmer.nl/raspbian/moonlight jessie main
#Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspbian.org/raspbian/ jessie main contrib non-free rpi
root@keppiePI:/home/dkeppens#

Also, have you run an apt-get update and apt-get upgrade to upgrade your system to the latest available packages before trying to add moonlight or are you running the default raspbian provided image unmodified ?

popcornkernaldontpanic, 2016-08-13:
I'm having the same problem. I've followed at least four different sites, re-wrote my entire Raspbian file system 3 times, and still always get hung up here. Did we ever get a resolution or a work around for this? Most recently, followed this one....

http://www.htpcbeginner.com/raspberry-pi-steam-box/

Yes, I checked multiple times for typos.

Droid22, 2016-09-02:
never mind i tried using raspian wheezy and it is working fine now

closed 2016-09-03 by irtimmer

wonderbai, 2017-07-29:
I have the exact same issue. is there a fix for this? or is the only solution to install wheezy instead of Jessie? would really like to keep my installation..