moonlight-stream/moonlight-embedded #383: Output audio to jack instead of HDMI

DarioRomano, 2016-06-15:
Please provide the following info.

**_NVidia Geforce Experience version:**
2.11.3.5
**Moonlight Embedded version:**
2.2.0
**Moonlight Embedded running on:**
 Raspberry Pi3 B

**Output of Moonlight Embedded:**
HDMI & Audio Jack

Hey, I just wanted to ask if there is a way to let Moonlight output audio via the Pis audio jack. I've only managed to get HDMI running, but since my Audio-Receiver does not support catching audio from HDMI, I'd have to input it with cinch. I got the cables and all, but can't seem to get the output away from HDMI out. I tried setting the standard output for audio of the Pi to jack using these steps: https://www.raspberrypi.org/documentation/configuration/audio-config.md .
Unfortunately Moonlight seems to have it's own configurations.

Thanks in advance!

7hunderbug, 2016-06-15:
Did you definitely run `amixer cset numid=3 1` and not `amixer cset 
numid=3 2` in the console? Or did you use `sudo raspi-config` to force 
the audio to the jack? Using amixer is a temporary fix, using 
raspi-config should edit your /boot/config.txt file making the change 
permanent across reboots.

If those don't work, you could try using the moonlight-embedded `-audio` 
switch to define which audio device to use. Off the top of my head, I 
think you can use `aplay -l` to list the audio devices to enable you to 
choose which one to use.

DarioRomano, 2016-06-15:
The raspberry is putting out sound from other applications through the jack. But once I start a stream it uses HDMI. I tried the -audio option, but it kept the stream from starting altogether. Maybe i input an invalid device. Anyway, I'll try the raspi-config soon and report back.

7hunderbug, 2016-06-15:
I'm not certain of this (I haven't got my moonlight Raspberry Pi running 
at the moment) but I think that moonlight defaults to using OMX for 
video and audio playback on the Raspberry Pi and therefore perhaps 
setting the audio device to 'local' might work, like this: `-audio 
local`. The reason I say this is because in the moonlight code, when OMX 
is used, the audio device defaults to 'hdmi' and according to the 
omxplayer documentation 
https://www.raspberrypi.org/documentation/usage/audio/, the OMX audio 
device is either 'hdmi' (HDMI audio) or 'local' (jack audio). This is 
also hinted at on lines 137-149 of src/audio/omx.c in the moonlight code.

DarioRomano, 2016-06-15:
> setting the audio device to 'local' might work, like this: `-audio
local`.
This solved the issue! Thank you very much!

7hunderbug, 2016-06-15:
Awesome :)

closed 2016-06-15 by irtimmer

Blaze99, 2017-05-08:
the "-audio local" managed to get it to the jack on the board. is it possible to direct it to a usb headset. so that i can use the mic as well?