moonlight-stream/moonlight-embedded #385: Will not pair. Complains about SSL certificate unagle to sign data since latest GXFE update

dkeppens, 2016-06-17:
Please provide the following info.

**_NVidia Geforce Experience version:_** : 2.11.3.5
**Moonlight Embedded version:** :  Moonlight Embedded 2.2.0 (latest master)
**Moonlight Embedded running on:** : Raspberry Pi 2

**Output of Moonlight Embedded:** : 

**First pairing attempt :** 

root@keppiePI:/home/dkeppens# moonlight pair 192.168.1.4
Moonlight Embedded 2.2.0 (EMBEDDED;PULSE;CEC;PI)
Connect to 192.168.1.4...
Generating certificate...done
NVIDIA GeForce GTX 760, GFE 2.11.3.5 (protocol version 7)
Please enter the following PIN on the target PC: 7832
1017 / 1017
**Failed to pair to server: Problem with the local SSL certificate**

**All subsequent pairing attempts :**

root@keppiePI:/home/dkeppens# moonlight pair 192.168.1.4
Moonlight Embedded 2.2.0 (EMBEDDED;PULSE;CEC;PI)
Connect to 192.168.1.4...
NVIDIA GeForce GTX 760, GFE 2.11.3.5 (protocol version 7)
Please enter the following PIN on the target PC: 9489
1017 / 1017
**Failed to pair to server: Failed to sign data**
root@keppiePI:/home/dkeppens# moonlight pair 192.168.1.4
Moonlight Embedded 2.2.0 (EMBEDDED;PULSE;CEC;PI)
Connect to 192.168.1.4...
NVIDIA GeForce GTX 760, GFE 2.11.3.5 (protocol version 7)
Please enter the following PIN on the target PC: 9795
1017 / 1017
**Failed to pair to server: Failed to sign data**
root@keppiePI:/home/dkeppens#

**What is the expected result?**

Should pair successfully with the PC

**What happens instead of that?**

See above. Complains about SSL certificate. It might be pertinent that GFXE just updated

dkeppens, 2016-06-21:
I tried a different approach and resolved this issue myself by manually uninstalling the compiled latest master version pulled from github and just adding the precompiled package from the repo mentioned in the wiki : 
deb http://archive.itimmer.nl/raspbian/moonlight jessie main
This works so i'll just close this issue. Beats me though as to why it wouldn't work when compiling myself from the latest master. I had a quick look in the code and as the error message also stated, it indicates a problem with the SSL certificate database. I'm not very familiar with BIO I/O programming, but i'm supposing the SSL certificate is just read from the /etc/ssl/certs certificate database directory. However, uninstalling and reinstalling the openssl and ca-certificates packages didn't help. I didn't check whether the relevant code was modified between the version of the precompiled package and the latest master since in this latest release the version number is still screwed up so it's hard to tell. Frankly though, it rather seems like something went wrong during the compilation process. My best guess is it's related to some library changes i may have pulled in before compiling when i added the following non-standard raspbian repo to be able to access a prepackaged version of kodi 16.1 (which is otherwise unavailable on raspbian, only serving 15.2 so far) which is built on pipplware (a different debian based distro) : 
deb http://pipplware.pplware.pt/pipplware/dists/jessie/main/binary /

So, in conclusion, anyone else running this setup : beware.

closed 2016-06-21 by dkeppens

christdu29, 2023-02-14:
Bonjours 
Comment associé ma tablette Samsung a t510 à mon PC  pour pouvoir connecté little navmaps  
En sachant que mon PC ne fait pas le WiFi 
Merci de m'aider