moonlight-stream/moonlight-embedded #389: App doens't launch

kristijanH, 2016-06-22:
NVidia Geforce Experience version: 2.11.4.0
Moonlight Embedded version: latest version (installed 21/6/2016)
Moonlight Embedded running on: Raspbbery Pi 2
Output of Moonlight Embedded: not sure

OS: OSMC May 2016 2016.05-1 (openelec version jessie)

Moonlight does not start. I tried the Chrome version and it works so the problem is on my Rasbbery device and not my PC.

I installed the jessie build of moonlight without any problems and paired it with my PC but when I try to start the app I get a messege:

Moonlight Embedded 2.2. (EMBEDDED;PI)
Too many options: No such file or directory

7hunderbug, 2016-06-22:
What command line are you using to launch moonlight?

kristijanH, 2016-06-22:
Normaly "moonlight streaming [] X.X.X.X". I also tried "sudo ..." with no effect. Tried also "moonlight streaming -app steam X.X.X.X".

I am using "putty" to input commands if that makes any difference.

7hunderbug, 2016-06-22:
Your syntax is incorrect. The correct syntax is:

moonlight stream X.X.X.X

I don't know where you got "streaming" from or "[]"

kristijanH, 2016-06-22:
Now it works! I got the instruction form here:

http://www.howtogeek.com/220969/turn-a-raspberry-pi-into-a-steam-machine-with-moonlight/

I gues the author missused the last command. Everything else was spot on.

Thank you for your time.

closed 2016-06-23 by kristijanH