moonlight-stream/moonlight-embedded #392: RTSP DESCRIBE request failed

StolenName, 2016-06-29:
Please provide the following info.

**NVidia Geforce Experience version: 2.11.4.0**
**Moonlight Embedded version: 2.2.0**
**Moonlight Embedded running on:**  Raspberry Pi 3

**Output of Moonlight Embedded:**
pi@rpn:~ $ moonlight stream
Moonlight Embedded 2.2.0 (EMBEDDED;PI)
Searching for server...
Connect to 192.168.2.100...
NVIDIA GeForce GTX 950, GFE 2.11.4.0 (protocol version 7)
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...Failed to receive RTSP reply payload
RTSP DESCRIBE request failed: -1
failed: -1
^CCleaning up platform...done

**What is the expected result?**
Stream showing on tv.

**What happens instead of that?**
nothing happens on tv.

Serethi, 2016-11-20:
I've had the same issue with Moonlight Embedded 2.2.2 and Nv GFE 3.1.0.52

Pairing works like a charm, also the Steam App or desired game starts on the PC 

I solved it by attaching the PI by wire and not WLAN, played some nice GTA V Hours after solving this ;)

DarioRomano, 2016-11-20:
Since mine is always connected by cable, that won't fix anything for me.

closed 2018-01-15 by StolenName