moonlight-stream/moonlight-embedded #394: Can't launch Games througth Steam

ElScratou, 2016-07-03:
Hey every one,

This is my configuration :
**_NVidia Geforce Experience version**: NVIDIA GeForce GTX 960, GFE 2.11.4.0 (protocol version 7)
**Moonlight Embedded version:** Moonlight Embedded 2.2.0 (EMBEDDED;PI)
**Moonlight Embedded running on:** Raspberry Pi 3
**Output of Moonlight Embedded:** TV Screen HDTV which support 1080p (tested with other devices)

**What is the expected result?**
I use the following commands to launch Steam with my 2 xbox 360 controllers. 
`sudo xboxdrv -i 0 -s & sudo xboxdrv -i 1 -s & moonlight stream -input /dev/input/event0 -input /dev/input/event1 -app Steam`

I can see the following sentences on my TV screen : 
```
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
```

Steam launch correctly on my TV screen. Only one of my controller is detected in Steam app (but that's not my major issue)

The result expected is when i clicked on "Launch" on my installed games, I can see it on my TV screen.

**What happens instead of that?**

The main issue is throught Steam when I want to launch one of my installed games. I click on "Launch" but no process or app seems to be launch on my computer. So after a while the screen reappers on steam and no games launch.

Thanks for helping me.

RealZack, 2016-07-04:
I have the same thing with mass effect,

try to manually add your game via preference, shield of nvidia geforce experience.

Then launch directly the game and bypass Steam

sudo xboxdrv -i 0 -s & sudo xboxdrv -i 1 -s & moonlight stream -input /dev/input/event0 -input /dev/input/event1 -app "your game"

Check with moonlight list if your game is well added.

ElScratou, 2016-07-06:
I've already done that with the nvidia geforce experience. We can see the 8 first games are those I had manually by nvidia geforce experience.
```
Moonlight Embedded 2.2.0 (EMBEDDED;PI)
Searching for server...
Connect to 192.168.1.15...
NVIDIA GeForce GTX 960, GFE 2.11.4.0 (protocol version 7)

1. Steam
2. LEGO_Star_Wars_Saga
3. LEGO_Movie
4. LEGO_Hobbit
5. LEGO_Harry_Potter
6. LEGO_Clone_Wars
7. LEGO_Batman
8. Fallout3
9. BioShock 2
10. Assassin's Creed III
11. Hearthstone: Heroes of Warcraft
12. The LEGO Movie - Videogame
13. Diablo III
14. BioShock 2
15. Fallout 3
```

When I launch you command, I bypass Steam, but I have the "nvidia stream" screen and nothing else than that :
```
Searching for server...
Connect to 192.168.1.15...
NVIDIA GeForce GTX 960, GFE 2.11.4.0 (protocol version 7)
Add input /dev/input/event0 (mapping /usr/share/moonlight/mappings/default.conf)...
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
```

I'm wondering if I have to do some port forwarding to permit computer <> pi to communicate ?

RealZack, 2016-07-06:
No port forwarding needed.

Try to connect a simple keyboard and launch Steam just with moonlight stream -app "Steam"

ElScratou, 2016-07-06:
I do more testing with your method.

It seems a majority of my games works like Borderlands, Fallout, Telltale games.

But I don't understand why all LEGO don't want to be launch .... I'm testing all those games and none of them are streamed.

The other games i can't launch are Blizzard games and Uplay games. I think the PI can't reach them because of the editor'apps.

RealZack, 2016-07-06:
I think it's a nvidia shield issues, I have the same problem with battlefield 4.

Not all the games are compatible

ElScratou, 2016-07-06:
Yeah but if the nvidia'list is good : https://shield.nvidia.com/game-stream

Some LEGO and Battlefield 4 are supported by the shield.

So the issue seems not come to nvidia.

Maybe steam issue, implementation of the moonlight ? or some parameters of your computer.

I'm surprised of issue on LEGO games because on other forums I can read some people who played LEGO specially Hobbit, Lord of the Ring. So I tested them and none of them works ....

I specially install this app to play with some friend so i'm a little disappointed to see other games work and not one those I want in particular.

ElScratou, 2016-07-25:
Anyone help ?

ElScratou, 2016-08-25:
I have this bug again ...

Bastbog, 2017-02-10:
I have the exact same issue, did you resolve yours?

juicypop, 2017-02-26:
I got the same issue with cracked games i added to steam library. I cracked my cd games so i didnt have to run each time i want to play another game to my PC and change the cd. None of the cracked games won't lunch through steam big picture.

Any update on this issue?

irtimmer, 2017-03-18:
This bug report looks like an issue in Steam and not in Moonlight Embedded.

closed 2017-03-18 by irtimmer