moonlight-stream/moonlight-embedded #397: Triying to compile in Orange Pi Plus 2

phyrexia, 2016-07-08:
im trying to install ins this new inexpensive and powerfull boards without much success.

Im using ARMBIAN , based on debian jessie

had to down the requirement for Cmake 3.1
root@orangepiplus:~/moonligth/moonlight-embedded/build# cmake ../
-- The C compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
CMake Error at CMakeLists.txt:2 (cmake_minimum_required):
  CMake 3.1 or higher is required.  You are running version 3.0.2

But I still have a lot of problems with dependencies, most of them could not be found.

```
root@orangepiplus:~/moonligth/moonlight-embedded/build# cmake ../
-- Found ALSA: /usr/lib/arm-linux-gnueabihf/libasound.so (found version "1.0.28") 
-- Found Opus: /usr/lib/arm-linux-gnueabihf/libopus.so  
-- Could NOT find Broadcom (missing:  BROADCOM_INCLUDE_DIR VCOS_LIBRARY VCHIQ_LIBRARY OPENMAXIL_LIBRARY BCM_HOST_LIBRARY) 
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY) 
-- Could NOT find Amlogic (missing:  AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY) 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for module 'libevdev'
--   found libevdev, version 1.3
-- checking for module 'libudev'
--   found libudev, version 215
-- checking for module 'sdl2>=2.0.4'
--   package 'sdl2>=2.0.4' not found
-- checking for module 'libavcodec'
--   package 'libavcodec' not found
-- checking for module 'libavutil'
--   package 'libavutil' not found
-- checking for module 'x11-xcb'
--   package 'x11-xcb' not found
-- checking for module 'vdpau'
--   package 'vdpau' not found
-- checking for module 'libpulse-simple'
--   found libpulse-simple, version 5.0
-- checking for module 'libcec>=3.0.0'
--   package 'libcec>=3.0.0' not found
CMake Error at CMakeLists.txt:52 (message):
  No video output available
```

```
root@orangepiplus:~/moonligth# cat /root/moonligth/moonlight-embedded/build/CMakeFiles/CMakeOutput.log
The system is: Linux - 3.4.112-sun8i - armv7l
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags: 

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/root/moonligth/moonlight-embedded/build/CMakeFiles/3.0.2/CompilerIdC/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /root/moonligth/moonlight-embedded/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec3830355114/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec3830355114.dir/build.make CMakeFiles/cmTryCompileExec3830355114.dir/build
make[1]: Entering directory '/root/moonligth/moonlight-embedded/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /root/moonligth/moonlight-embedded/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec3830355114.dir/testCCompiler.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec3830355114.dir/testCCompiler.c.o   -c /root/moonligth/moonlight-embedded/build/CMakeFiles/CMakeTmp/testCCompiler.c
[... 85 lines cut ...]
    arg [-lgcc] ==> lib [gcc]
    arg [--as-needed] ==> ignore
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [--no-as-needed] ==> ignore
    arg [-lc] ==> lib [c]
    arg [-lgcc] ==> lib [gcc]
    arg [--as-needed] ==> ignore
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [--no-as-needed] ==> ignore
    arg [/usr/lib/gcc/arm-linux-gnueabihf/4.9/crtend.o] ==> ignore
    arg [/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/crtn.o] ==> ignore
  remove lib [gcc]
  remove lib [gcc_s]
  remove lib [gcc]
  remove lib [gcc_s]
  collapse library dir [/usr/lib/gcc/arm-linux-gnueabihf/4.9] ==> [/usr/lib/gcc/arm-linux-gnueabihf/4.9]
  collapse library dir [/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf] ==> [/usr/lib/arm-linux-gnueabihf]
  collapse library dir [/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../..] ==> [/usr/lib]
  collapse library dir [/lib/arm-linux-gnueabihf] ==> [/lib/arm-linux-gnueabihf]
  collapse library dir [/usr/lib/arm-linux-gnueabihf] ==> [/usr/lib/arm-linux-gnueabihf]
  implicit libs: [c]
  implicit dirs: [/usr/lib/gcc/arm-linux-gnueabihf/4.9;/usr/lib/arm-linux-gnueabihf;/usr/lib;/lib/arm-linux-gnueabihf]
  implicit fwks: []

```

7hunderbug, 2016-07-08:
"(message): No video output available" should give you a hint - Orange 
Pi isn't supported "out of the box" by moonlight.
At a bare minimum, to get moonlight to compile, you need supported 
video-decoder development libraries installed - currently moonlight 
supports the following VPUs and libraries: Broadcom OpenMAX IL 
(Raspberry Pi), Freescale IMX, Amlogic (ODROID C1) and also, I believe, 
software decoding using ffmpeg/SDL2/VDpau.

Long story short: you are probably going to have to do some hacking 
and/or coding to get it to work.

Here are some links you might find helpful:

https://github.com/orangepi-xunlong - vdpau and cedarx libraries for 
Orange Pi

http://linux-sunxi.org/Cedrus/libvdpau-sunxi - wiki regarding these 
libraries

http://forum.odroid.com/viewtopic.php?f=91&t=15456 - forum regarding 
compiling moonlight to run on ODROID U2/U3 using ffmpeg and SDL software 
decoding

http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=723 - 
some information regarding Orange Pi Cedar / VDPAU video acceleration 
libraries.

You might try the ffmpeg/SDL/OpenGLES route or you might try the 
ffmpeg/VDPAU route.

Good luck!

irtimmer, 2016-12-27:
Duplicate of #249

closed 2016-12-27 by irtimmer