moonlight-stream/moonlight-embedded #398: Input Lag at 60fps/1080p on raspberry pi 2

shayaantx, 2016-07-22:
Please provide the following info.

**_NVidia Geforce Experience version:_** 2.11.4.0
**Moonlight Embedded version:** 2.2.1
**Moonlight Embedded running on:** _Raspberry Pi 2

Hello,

I'm on latest gfe server version and latest moonlight embedded version on my pi. I can't seem to avoid input lag at 1080/60fps. The stream quality looks great, but there is noticeable input lag. I do NOT see any input lag at 1080/30fps or 720/60fps, both the stream and audio and input are perfect.

I enabled force_turbo in the boot/config.txt, but that didn't seem to make it better.

Any ideas?

Thanks

sc-o, 2016-09-08:
A few ideas. I have mine running at 1080/60fps decently well at the moment and I did a few things that might help, though I admit that some of it is a little unscientific:

Make sure you have a decent power supply, I was running moonlight initially on a 5V 1A supply, which worked at 720p, but got horrible input lag when I scaled up. The pi was starved for power and was throttling itself to keep from crashing. I migrated to a 2.5A supply which helped a lot.

Overclock the GPU. I believe force turbo only works on the CPU. You can overclock the entire GPU at once, or do it in parts. I haven't tested this extensively, but it would seem that overclocking the h.264 decoder independently might give better results and you should be able to overclock it higher than you would the entire GPU. YMMV, and a heatsink would help. Word also is that you can overclock higher and risk less file corruption if you boot to a USB flash drive. I'm running at a 300 MHz Core clock and 350 MHz h.264 clock on my Pi2 and could probably go a little higher.

Assign more memory to the GPU. The GPU and CPU on the pi share memory and the GPU is given very little. I upped mine to about 300 MB.

I'm a little unsure about how impactful the GPU overclocking and memory alocation is. I'd really like to try to benchmark it somehow one of these days to figure out how to get the best quality stream out of the device.

cgutman, 2016-09-08:
force_turbo alone isn't good enough. You need:

force_turbo=1
h264_freq=333

shayaantx, 2016-09-10:
I'll try out your guys suggestions. Only reason I didn't use "h264_freq" was cause I heard it voids the warranty.

Thanks for the responses

closed 2017-03-18 by irtimmer

Jayatubi, 2018-05-05:
@cgutman You're the life saver!