moonlight-stream/moonlight-embedded #399: moonlight embedded raspberry pi 3 Gforce 2.11.4.0 moonlight 2.2.1 Fails to pair with pc

collier45, 2016-07-24:
Please provide the following info.

**_NVidia Geforce Experience version:_**
**Moonlight Embedded version:**
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._

**Output of Moonlight Embedded:**

**What is the expected result?**

**What happens instead of that?**

collier45, 2016-07-24:
every time trying to pair i am left with 

Moonlight Embedded 2.2.0 (EMBEDDED;PI)
Connect to 192.168.1.6...
NVIDIA GeForce GTX 960, GFE 2.11.4.0 (protocol version 7)
Please enter the following PIN on the target PC: 5712
Failed to pair to server: Pairing failed

joffreydupire, 2016-08-30:
I've the same issue, did you find something ?

joffreydupire, 2016-08-30:
Are you using remote desktop of windows ?

closed 2017-03-18 by irtimmer