moonlight-stream/moonlight-embedded #400: Video not working

Bl4cke, 2016-07-25:
Please provide the following info.

**_NVidia Geforce Experience version:2.11.4.0_**
**Moonlight Embedded version:2.2.1**
**Moonlight Embedded running on:** _Raspberry Pi_

**Output of Moonlight Embedded:**
Audio
**What is the expected result?**
The pi should output the video and the audio of the game.
**What happens instead of that?**
The computer receives input from the controller on the Pi, the audio plays just fine (with some latency) but instead of outputting video it gets stuck on emulation station.

BadFlo, 2016-10-05:
Same issue here, except that mine shows command line with "Waiting for IDR frame" over and over again. Input is fine though (moving on my computer).

cgutman, 2016-10-06:
@BadFlo Can you file a separate issue? This issue is specific to GeForce Experience 3.0.7.

Tim-ofzo, 2016-11-24:
I have the same problem with the Model B 512MB with both the latest version of GFE and Moonlight

DTibbs, 2017-01-21:
Ran across this issue on my PI 3 Model B. I noticed I had the experimental OpenGL driver on in the raspi-config. Turning off the experimental OpenGL driver, and my video works again.

closed 2017-03-18 by irtimmer