moonlight-stream/moonlight-embedded #401: Moonlight acts like a controller (and there is no way to disable it)

kartercas, 2016-07-25:
**_NVidia Geforce Experience version:_2.11.4.0**
**Moonlight Embedded version: 2.2.0**
**Moonlight Embedded running on:** _Raspberry Pi 2..._

Because Resbian doesn't work (good) with the Xbox One controller, I have decided to get a long usb cable so I can connect it to my computer natively.

But Moonlight already acts like a controller, so now when I start moonlight (with steam BIG pic mode) I says that there are 2 controllers when there is only one.

My plugged in pc controller is also the second controller, and that makes some games unplayable.

Can we have a option to disable the input in a future update?

cgutman, 2016-07-27:
There is a way you can disable remote gamepad input on your PC, see here: https://github.com/moonlight-stream/moonlight-docs/wiki/Setup-Guide#using-a-gamepad-connected-to-the-pc-instead-of-the-streaming-device

But I'm also curious, what issues did you have with the Xbox One controller on your Pi? It's working fine for me.

kartercas, 2016-08-01:
Thanks for sharing!

About the Xbox one controller (not moonlight related)
I am using rasbian jessie, and download Xinput (or something like that). My 360 controller works "perfect", and my One controller doesn't get recognized.

But (moonlight related) I found that using a 15 meter usb cable is more responsive than using a controller with moonlight.

Direcka, 2016-10-12:
@cgutman Renaming dll files (in both folders) is not working for me :( i have still 2 xinput controllers in steam and canot play some games.

closed 2017-03-18 by irtimmer