moonlight-stream/moonlight-embedded #410: Quit while running

cain4355, 2016-09-10:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.0.5.22
**Moonlight Embedded version:** 2.2.0
**Moonlight Embedded running on:** Raspberry Pi

**Output of Moonlight Embedded:**
Moonlight Embedded 2.2.0 (EMBEDDED;PI)
Searching for server...
Connect to 192.168.0.2
NVIDIA GeForce GTX 960, GFE 3.0.5.22 (protocol version 7)

**What is the expected result?** Application exit

**What happens instead of that?** Nothing

When attempting to quit an application while Moonlight is running nothing happens. I run the "moonlight quit" command and the application does not exit. I have a python script listening for my controller inputs and when I press the XBOX button on the controller it executes the aforementioned command. I see that moonlight attempted to quit. However, this does not kill the moonlight stream. Both the hosting PC and moonlight are still running the game.

cain4355, 2016-09-14:
Solved my problem. When executing the action from a controller it was running it with sudo privileges. By forcing it to run the command as user pi it killed the stream perfectly.

closed 2016-09-14 by cain4355

mercier1200, 2016-09-16:
Hi cain4355,
Just by curiosity, are you using the python utility inputexec to achieve that?

If yes, I am using this piece of software as well but ran into the same problem than you. How can you force the command "moonlight quit" to run as user pi like you said?

Thanks in advance for your answer.

cain4355, 2016-09-16:
The python script I used was designed to read directly from xboxdrv. https://github.com/martinohanlon/XboxController

However, I decided to go a different route. I instead mapped the xboxdrv button directly using the xboxdrv instead of having to use a python script and xboxdrv. This mapped button calls a sh scirpt that kills all moonlight processes on the pi and then runs: sudo -u pi moonlight quit &.

So now after force quitting the application on the pi it kills the moonlight stream. The reason I did this is I had problems killing the stream then the application and still having the Nvidia Game Stream screen on the pi.

mercier1200, 2016-09-16:
Ok thanks for your input,

Since xboxdrv did not work great on raspbian Jessie, I switched to xpad instead. I think I will try xboxdrv again to do something like you. I suppose you use a specific parameter when starting xboxdrv, like a config file to override the default configuration map?

mercier1200, 2016-09-16:
Never mind, you probably used the _exec_ parameter.