moonlight-stream/moonlight-embedded #412: moonlight acting as if not paired

Stevenisnormal, 2016-09-18:
Please provide the following info.

GFE version 3.0.5.22
Moonlight Embedded 2.1.4
Running on RPi 2

pi@retropie:~/RetroPie/roms/moonlight $ moonlight pair 192.168.0.5
Moonlight Embedded 2.1.4 (EMBEDDED;PULSE;PI)
Connect to 192.168.0.5...
Please enter the following PIN on the target PC: 7343
Failed to pair to server: Already paired

After I entered this:
moonlight pair 192.168.0.5
Moonlight Embedded 2.1.4 (EMBEDDED;PULSE;PI)
Connect to 192.168.0.5...
Please enter the following PIN on the target PC: 6304
1017 / 1017
Succesfully paired

I want to just be able to stream, somehow.

But, It tells me to pair when I am already paired.

wackerl91, 2016-09-18:
@Stevenisnormal 
Have you tried using the [latest version](https://github.com/irtimmer/moonlight-embedded/releases) of Moonlight Embedded (2.2.1)? As of 2.2.0 it contains support for GFE 2.11 and as far as I know the protocol hasn't changed between GFE 2.11 and GFE 3.x (at least so far), so your current version of Moonlight is most likely incompatible to your version of GFE.

Stevenisnormal, 2016-09-18:
Thank you! That works much better, I guess I just wasn't thinking about the version!

closed 2016-09-18 by Stevenisnormal