moonlight-stream/moonlight-embedded #415: No video with GeForce Experience 3.0.7

cgutman, 2016-10-06:
NVIDIA made a breaking change in the video frame header format in GeForce Experience 3.0.7 that causes devices to fail to decode the video stream.

I have fixed the issue upstream in moonlight-common-c: https://github.com/moonlight-stream/moonlight-common-c/commit/bd825776b3d40ea834bac2352e4359bb3145b367

But it comes with a breaking API change that needs some slight modifications to the client to directly pass the 'appversion' string unmodified from the serverinfo query rather than just the major version number.

You may find these changes helpful to show how Moonlight iOS and Chrome were updated to the newer API:
https://github.com/moonlight-stream/moonlight-ios/commit/23564a3028d39306f57d37bc05e673e6345eb18e
https://github.com/moonlight-stream/moonlight-chrome/commit/ee5ee588bc1bfe226e911d4e09c2a065be53b1e2

Armzeruel, 2016-10-08:
pls teach me how to make stream work with 3.0.7, dont understand code and what's in fixes

irtimmer, 2016-10-08:
Fixed in release 2.2.2

closed 2016-10-08 by irtimmer

Armzeruel, 2016-10-08:
im sorry but 2.2.2 dont have vpk vita file, so im clueless about how to
implement it