moonlight-stream/moonlight-embedded #419: Map pad wired xbox360 - little problem

Kheiryk, 2016-10-08:
Hi all,
When i use my xbox.map, button RT / LT not working
When i don't use my xbox.map (when i start a game in the bash file), X and Y button are inverse

Do you know how i can resolve it please ?
thanks

xbox.map 
```
abs_x = 0
abs_y = 1
abs_z = 10
abs_rx = 3
abs_ry = 4
abs_rz = 9
abs_deadzone = 0
abs_dpad_x = 17
abs_dpad_y = 16
reverse_x = false
reverse_y = true
reverse_rx = false
reverse_ry = true
reverse_dpad_x = false
reverse_dpad_y = false
btn_east = 305
btn_south = 304
btn_north = 308
btn_west = 307
btn_select = 314
btn_start = 315
btn_mode = 316
btn_thumbl = 317
btn_thumbr = 318
btn_tl = 310
btn_tr = 311
btn_tl2 = 312
btn_tr2 = 313
btn_dpad_up = -1
btn_dpad_down = -1
btn_dpad_left = -1
btn_dpad_right = -1

```

CUD, 2016-10-16:
I'm having the same issue with the X and Y buttons being switched. I tried entering the inputs in reverse when prompted (X when prompted for Y and vice versa) but it still didn't work, strangely enough.

Athou, 2017-01-08:
Had the same issue, I played a little bit with the mappings and I think I finally have something working for the wired xbox controller:

```
abs_x = 0
abs_y = 1
abs_z = 2
reverse_x = false
reverse_y = true
abs_rx = 3
abs_ry = 4
abs_rz = 5
reverse_rx = false
reverse_ry = true
abs_deadzone = 0
abs_dpad_x = 16
abs_dpad_y = 17
reverse_dpad_x = false
reverse_dpad_y = false
btn_north = 308
btn_east = 305
btn_south = 304
btn_west = 307
btn_select = 314
btn_start = 315
btn_mode = 316
btn_thumbl = 317
btn_thumbr = 318
btn_tl = 310
btn_tr = 311
btn_tl2 = -1
btn_tr2 = -1
btn_dpad_up = 546
btn_dpad_down = 547
btn_dpad_left = 544
btn_dpad_right = 545
```

closed 2017-03-18 by irtimmer