moonlight-stream/moonlight-embedded #420: Moonlight will not Pair

fil5000, 2016-10-09:
Please provide the following info.

**_NVidia Geforce Experience version:_**
3.0.7.34
**Moonlight Embedded version:**
2.2.2
**Moonlight Embedded running on:** Raspberry Pi

**Output of Moonlight Embedded:**

pi@raspberrypi:~ $ moonlight pair 192.168.0.108
Moonlight Embedded 2.2.2 (EMBEDDED;PI)
Connect to 192.168.0.108...
NVIDIA GeForce GTX 970, GFE 3.0.7.34 (protocol version 7)
Please enter the following PIN on the target PC: 6264
Failed to pair to server: Pairing failed

**What is the expected result?**
Should be pairing with PC
**What happens instead of that?**
Just the error message

Moonlight was working fine up until I installed the latest GeForce experience and display drivers a couple of days ago. I'm running Windows 10 if that makes a difference.

Levitas, 2016-10-10:
Hi, i had the same problem.

Try reinstalling Geforce, this worked for me (for the pairing.. Streaming still doesnt work..)

irtimmer, 2016-10-10:
Duplicate of #399

closed 2016-10-10 by irtimmer