moonlight-stream/moonlight-embedded #422: Stram starts and disconnects after 3 - 5 seconds

Levitas, 2016-10-10:
Please provide the following info.

**_NVidia Geforce Experience version: NVIDIA GeForce GTX 660 Ti, GFE 3.0.7.34 (protocol version 7)**
**Moonlight Embedded running on:** Moonlight Embedded 2.2.2 (EMBEDDED;PI) [OSMC]

**Output of Moonlight Embedded:**
root@rpi:/# moonlight stream 192.168.0.104
Moonlight Embedded 2.2.2 (EMBEDDED;PI)
Connect to 192.168.0.104...
NVIDIA GeForce GTX 660 Ti, GFE 3.0.7.34 (protocol version 7)
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
EVIOCGRAB failed with error 16
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Terminating RTSP handshake...done
Cleaning up platform...done
root@rpi:/# EVIOCGRAB failed with error 16Loss Stats: Transaction failed: 11
-bash: EVIOCGRAB: command not found

**What is the expected result?**
See the video and play :D it closes after a couple of seconds (1 - 5)

Levitas, 2016-10-10:
After restarting pi and pc i get the following output:
root@rpi:~# moonlight -app "mafia3" stream 192.168.0.104
Moonlight Embedded 2.2.2 (EMBEDDED;PI)
Connect to 192.168.0.104...
NVIDIA GeForce GTX 660 Ti, GFE 3.0.7.34 (protocol version 7)
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...RTSP: Failed to connect to UDP port 48010
failed: -1
EVIOCGRAB failed with error 16
^CCleaning up platform...done

No Firewall, no antivirus or sth like that

irtimmer, 2016-10-10:
Duplicate of #368

closed 2016-10-10 by irtimmer