moonlight-stream/moonlight-embedded #423: RTSP Failed to connect to UDP port 48010 failed:-1

blast1987, 2016-10-12:
Hello i am getting this error on GFE 3.0.7: RTSP Failed to connect to UDP port 48010 failed:-1

GFE version 3.0.7.34
moonlight embedded version 2.2.2
Device: Raspberry Pi 3

georgcmef, 2016-10-13:
I get the same error, a workaround: reinstalling Geforce experience after every system POST works till the next POST.

blast1987, 2016-10-13:
So does it mean i have to reinstall GFE after every reboot?

raynetec, 2016-10-14:
There's a problem in the latest GFE 3.0.7.34. Try to lock and then unlock your system and see if that works for you.
Source: [GeForce Forum](https://forums.geforce.com/default/topic/970697/gamestream/373-06-driver-completely-broke-gamestream-error-8033020f-when-trying-to-launch-a-game-/post/4993302/#4993302)

Gobol, 2016-10-24:
I got this error when I request incompatible resolution with my primary server display...

closed 2017-03-18 by irtimmer