moonlight-stream/moonlight-embedded #424: No Video, "Waiting for IDR frame" loop on GFE 3.0.7

robmcgrath, 2016-10-16:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.0.7.34
**Moonlight Embedded version:**2.2.0
**Moonlight Embedded running on:** Raspberry Pi 2

**Output of Moonlight Embedded:**
Audio: OK
Keyboard input: OK
Video: does not start

**What is the expected result?**
Video streaming begins.

**What happens instead of that?**
Waiting for IDR frame (repeats many times)
Reached consecutive drop limit
IDR frame request sent
Waiting for IDR frame
(repeats until interrupted)

cgutman, 2016-10-16:
This is fixed by Moonlight Embedded 2.2.2.

robmcgrath, 2016-10-16:
Are there any plans to update the packages at http://archive.itimmer.nl/raspbian/moonlight to 2.2.2?

irtimmer, 2016-10-16:
Version 2.2.2 is released for Raspbian Wheezy

closed 2016-10-16 by irtimmer