moonlight-stream/moonlight-embedded #427: Controller not working in some instances.

sirus887, 2016-10-26:
**_NVidia Geforce Experience version: 3.0.7.34
**Moonlight Embedded version: 2.2.2 (I think)
**Moonlight Embedded running on: Raspberry Pi 2

I am having some problem with controllers for certain games. I have two identical pc xbox 360 controller and have created the map file.  I have no problem using both controllers for most of my games but in some instances the controller doesnt work at all or the 2nd controller is not recognised.

Game 1 - Pro Evo 2016
The controller doesnt work at all. However, playing directly from my pc, the controller is fine.

Game 2 - Rayman Origins/Legends
The second controller doesnt work.

I have tried other 2 players games such as the Lego franchise, and both controllers work fine. 
Just curious if anyone has had this issue? 

My moonlight cmd is:
moonlight stream 192.168.0.101 -1080 -60fps -mapping /opt/retropie/configs/moonlight/xbox.map

DarioRomano, 2016-11-03:
Having a similar problem. Started for me with the switch to version 3 of GFE. Sadly the number of nonfunctional games seems to be rather big for me. Haven't found a solution other than downgrading GFE yet.

sirus887, 2016-11-03:
I have tried various versions of GFE, all having the same result. @DarioRomano What version of GFE are you using?

DarioRomano, 2016-11-03:
2.11.3.5 was definitely working flawlessly for me. I think they might have done further updates until they released GFE 3. The issues only started with the newest major update of GFE.

kriegalex, 2016-11-05:
Just uninstalled 3.1.0.52 to try with old one, in my case didn't change anything. I have to point out that I was using the controller on the host PC, so I moved it to the client and then it worked. If your controller is already attached to the client, then it's really annoying...

sirus887, 2016-11-07:
@DarioRomano  Tried that version of GFE but to no avail. :(

@kriegalex  Are you saying your method was to put the controllers into your pc, then once the game starts, move them to the pi?

irtimmer, 2017-03-18:
Bug looks like an issue in Geforce Experience and not in Moonlight Embedded

closed 2017-03-18 by irtimmer