moonlight-stream/moonlight-embedded #429: Moonlight Not Paring With PC

craftyjai, 2016-10-29:
When I try to pair with my pc it only says "Failed to pair to server: Pairing failed"
I do moonlight pair 192.168.0.199 and it only says failed

`root@raspberrypi:/usr/local/bin# moonlight pair 192.168.0.199
Moonlight Embedded 2.2.2 (EMBEDDED;PI)
Connect to 192.168.0.199...
NVIDIA GeForce GTX 1070, GFE 3.0.7.34 (protocol version 7)
Please enter the following PIN on the target PC: 9470
Failed to pair to server: Pairing failed
root@raspberrypi:/usr/local/bin#`

GFE Version: 3.0.7.34
Device: Raspberry Pi 2

cgutman, 2016-10-30:
Download the latest driver for your GPU from NVIDIA's website. It comes with a newer version of GFE (3.1.0.52) which should fix this bug.

craftyjai, 2016-10-30:
It worked

closed 2016-10-30 by craftyjai