moonlight-stream/moonlight-embedded #431: Audio Output on both host and client?

wilkie1020, 2016-11-03:
Please provide the following info.

**_NVidia Geforce Experience version:_**
3.0.7.34
**Moonlight Embedded version:**
2.2.2
**Moonlight Embedded running on:** 
Raspberry Pi 3
**What is the expected result?**
Would like to have audio on both simultaneously so I can use one as a watching area.

**What happens instead of that?**
Everything works fine using audio on one device versus the other and I can very easily set default in sound options to switch it to and from. 

Does anyone know of a reliable method to output audio to both host and client simultaneously in a fashion such that one can watch someone play (Say, VR) from the safety of another room. My set up works fine and streams reliably with(virtually) no stutter through a Gigabit switch all on CAT6 to an RPi 3

Thank you,

wilkie1020, 2016-11-04:
I believe I found a solution using stereo mix in the Windows sound settings. Mild delay, but might do the trick.

stuarta0, 2017-02-03:
I second this feature. My use case is streaming to the lounge to play games whilst streaming the same content over RTMP to the interwebs.

irtimmer, 2017-03-18:
This use case isn't supported by GameStream. Therefore it can't be implemented in Moonlight Embedded.

closed 2017-03-18 by irtimmer

Spamm00r, 2018-07-13:
This works on the Iphone app, but not on Raspberry as a client. 

I'd like to always have that the sound also plays on the host. The setting in luna launcher has no effect. Unfortunately sometimes after the moonlight session ends, the sound does not resume on the host.

It would eb better to have always sound running also on the host.