moonlight-stream/moonlight-embedded #432: Raspberry Pi 3 and PC won't pair.  PC never shows PIN to enter.

goodpunk6, 2016-11-03:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.0.7.34
**Moonlight Embedded version:** 2.2.2
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._  Raspberry Pi 3

**Output of Moonlight Embedded:**  just says connecting and the IP address.  then it says it fails to pair

**What is the expected result?**  I expect my PC to prompt me to pair.  My PC is logged into Nvidia and ready to pair.

**What happens instead of that?**  Nothing.  I never get prompted for anything.

wilkie1020, 2016-11-03:
Game streaming enabled in Geforce Experience?

goodpunk6, 2016-11-03:
Yes it is enabled on the PC

cgutman, 2016-11-03:
Try upgrading to GeForce Experience 3.1.0.52

goodpunk6, 2016-11-03:
I will upgrade when I get home.  I'll report back then.

goodpunk6, 2016-11-05:
I upgraded Geforce Experience!  It finally prompted me for the pin!  However, it won't start streaming.

goodpunk6, 2016-11-05:
correction, i reran the command, it works!  yessssss!!!  i was streaming Gears of War 4!!!  I can't wait to give this as a gift for christmas!

goodpunk6, 2016-11-05:
Ok, I still have a problem...no sound.  =( where are the logs?  I'd love to post/share them.

closed 2016-11-05 by goodpunk6