moonlight-stream/moonlight-embedded #434: Video works but No Audio.  Audio and GPU drivers are the latest

goodpunk6, 2016-11-07:
Please provide the following info.

**_NVidia Geforce Experience version:_**  3.1.0.52
**Moonlight Embedded version:**  2.2.2
**Moonlight Embedded running on:** Raspberry Pi 3

**Output of Moonlight Embedded:**  
 moonlight stream 192.168.1.43
Moonlight Embedded 2.2.2 (EMBEDDED;PI)
Connect to 192.168.1.43...
NVIDIA GeForce GTX 970, GFE 3.1.0.52 (protocol version 7)
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done

output of aplay -l 
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

**What is the expected result?**
To be able to hear audio while streaming

**What happens instead of that?**

Video starts but there is no audio.  I have updated the Geforce experience driver and update my audio driver to the latest.

Bernatbali, 2017-01-17:
Hello,

I have a same problem. start the moonlight with:
moonlight steram -app Stream
and the I have video but no sound.
somebody has any idea?

irtimmer, 2017-03-18:
Select correct audio output HDMI/analog using `-audio` option

closed 2017-03-18 by irtimmer