moonlight-stream/moonlight-embedded #435: can't pair 2.2.2 with GFE 3.1.0.52

grgohr, 2016-11-11:
**GFE 3.1.0.52
Moonlight Embedded 2.2.2 (SDL;CEC) running on Odroid XU4**

**Output of Moonlight Embedded:**
```
# moonlight pair
Moonlight Embedded 2.2.2 (SDL;CEC)
Searching for server...
Connect to 192.168.0.2...
Can't connect to server 192.168.0.2
```

**What is the expected result?**
Gamestream is on
http://127.0.0.1:47989/serverinfo?uniqueid=1234 gives a proper XML
the PIN should be requested in GFE

**What happens instead of that?**
=> the PIN request never shows under GFE
=> the keydir option is ignored, a folder "192.168.0.2" is automatically created instead (with usual key files inside despite the missing PIN request step)
=> nothing in GFE shows an existing shield peripheral

grgohr, 2016-11-15:
better with the latest GFE update but still nowhere to input the PIN 

`# moonlight pair`
`Moonlight Embedded 2.2.2 (SDL;CEC)`
`Searching for server...`
`Connect to 192.168.0.2...`
`Generating certificate...done`
`NVIDIA GeForce GTX 970, GFE 3.1.0.52 (protocol version 7)`
`Please enter the following PIN on the target PC: 2136`
`Failed to pair to server: Pairing failed`

romanovzky, 2016-11-16:
Luckier than me, mine doesn't even go through Connect to xxx.xxx.xxx.xxx

Any ideas?

grgohr, 2016-11-16:
have you triggered the gamestream option in GFE prefs ?
is http://127.0.0.1:47989/serverinfo?uniqueid=1234 returning an XML ?

romanovzky, 2016-11-17:
In this new GFE I can't seem to interact with any gamestream option. It just says "ready", but there's no switch or button. In GF2 there were options, list of paired devices, etc in this one is just another listing in the settings, but not interactive.

The second link does nothing, as clicked here. Should I change it to fit my own network IPs? I don't understand what that link should do...

romanovzky, 2016-11-17:
This is what I see on the PC GFE

[image]

cgutman, 2016-11-17:
There should be a SHIELD tab on the left side of the settings page. That's where you enable GameStream.

romanovzky, 2016-11-17:
This is my settings, no SHIELD tab or option. Is this normal?

[image]

cgutman, 2016-11-17:
No, but you can try enabling "Allow experimental features", wait 5 minutes or so, then restart GeForce Experience and you should get an update. See if that update fixes your issue.

grgohr, 2016-11-17:
and if not appearing you should uninstall/reinstall, the shield tab should appear on the screen.

now back to the original issue of the thread, any idea why the PIN popup does not show up while pairing ?

grgohr, 2016-11-17:
mmh actually followed my own advice : uninstalled GFE, reinstalled.
now the popup appears when pairing.
GFE is pretty bugged when upgraded...

closed 2016-11-17 by grgohr

romanovzky, 2016-11-18:
My issue was also solved, thanks guys!