moonlight-stream/moonlight-embedded #436: "Waiting for IDR Frame" Issue

Burgeh, 2016-11-17:
**_NVidia Geforce Experience version:_** 3.1.0.52
**Moonlight Embedded version:** I am assuming latest since I did an apt-get update but I installed this on my RPI 2 https://github.com/aikoncwd/aikoncwd-rpi-mediacenter
**Moonlight Embedded running on:** _Raspberry Pi 2

**Output of Moonlight Embedded:**
Just a spam of "Waiting for IDR Frame"

**What is the expected result?**
Steam would be visible on my Pi

**What happens instead of that?**
Just the terminal spamming "Waiting for IDR Frame" 
I can control my remote computer with my controller from the Pi but no video or audio.

NoDuNT, 2016-11-17:
Hi !
I have the same issue. Spam of "Waiting for IDR frame" with some "Reached consecutive drop limit" followed by "IDR frame request sent"

I'm on Windows 10 + Raspberry Pi 3

I can't see my desktop but I can control my PC throught the Pi with mouse and keyboard
I think it happened after I updated my Nvidia drivers and thus Geforce Experience. 

I tried to come back to the 3.0.7.34 version of GFE but I still have the issue.

Burgeh, 2016-11-17:
Mine also says something similar to that in between the spam. It just moves so fast I can't read it haha. Is there more information I should provide than this, and do you think downgrading Geforce Experience would work?

cgutman, 2016-11-17:
You need to upgrade to Moonlight Embedded 2.2.2 for support for GFE 3.0.7 and above.

If you installed using the instructions on this wiki, you should be able to do that via:
sudo apt-get update
sudo apt-get upgrade

Burgeh, 2016-11-17:
Ah, I see. Thanks a ton actually.

NoDuNT, 2016-11-19:
I feel dumb I didn't even think there could be a moonlight update. Thanks it works just fine now !

craptec, 2016-12-12:
> You need to upgrade to Moonlight Embedded 2.2.2 for support for GFE 3.0.7 and above.
> If you installed using the instructions on this wiki, you should be able to do that via:
> sudo apt-get update
> sudo apt-get upgrade

I tried this several times.  It does not update moonlight. I'm still on 2.2.0

Running Raspian Jessie on RPi2. 
`deb http://archive.itimmer.nl/raspbian/moonlight jessie main` is added to /etc/apt/sources.list
When running sudo apt-get install moonlight-embedded it says "moonlight-embedded is already the newest version"

Help will be highly appreciated =)

closed 2016-12-12 by irtimmer

chibchibi, 2016-12-16:
Hey , have same issue here

Moonlight Version 2.2.2 
Geforce Experience 3.2.0.96
(gtx 970)

Audio and Control Stream Work fine..
But no video stream works  just spamms Waiting for IDR frame in shell ..

i already clean installed geforce experince and moonlight  several times

any ideas ?

leizzer, 2016-12-18:
Same issue here with Raspberry 2