moonlight-stream/moonlight-embedded #437: controller not working in steam

grgohr, 2016-11-17:
**GFE 3.1.0.52**
**Moonlight Embedded version: 2.2.2**
**Moonlight Embedded running on: Odroid XU4**

Despite a proper moonlight pairing + mapping setup done on the remote control (Xin-Mo Dual arcade sticks) and the following proper launch command using the mapping file generated:

`moonlight stream -remote -keydir /recalbox/share/system/configs/moonlight/keydir -720 -60fps -mapping /recalbox/share/system/configs/moonlight/mapping.conf -input /dev/input/event0 -mapping /recalbox/share/system/configs/moonlight/mapping.conf -input /dev/input/event1`

Steam big picture launches but does not react to the remote arcade controller

forced to use the keyboard on host pc to close steam, no game is playable due to that issue

same if only one controller is specified or none
`moonlight stream -remote -keydir /recalbox/share/system/configs/moonlight/keydir -720 -60fps -mapping /recalbox/share/system/configs/moonlight/mapping.conf`

grgohr, 2016-11-26:
not only the Xinmo, but the iBuffalo as well is not working at all

irtimmer, 2017-03-18:
Bug report misses output of Moonlight Embedded

grgohr, 2017-03-20:
maybe this can help ? 
the controller is detected properly when executing "moonlight map" but does not react when streaming

# moonlight stream -app Steam
Moonlight Embedded 2.2.2 (SDL;CEC)
Searching for server...
Connect to 192.168.1.2...
NVIDIA GeForce GTX 970, GFE 3.4.0.70 (protocol version 7)
sh: setterm: command not found
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...Failed to open audio: ALSA: Couldn't open audio device: Device or resource busy
done
Starting input stream...done
Decode failed - Invalid data found when processing input
Decode failed - Invalid data found when processing input

irtimmer, 2017-03-21:
Yes, this output is very helpful. When you use SDL (required as your platform doesn't support hardware acceleration) the SDL api is used to get input from controllers and therefore the -input and -mapping options are ignored. So currently you have to search on how to get your gamepad configured for SDL2 to get it working. Maybe I will try to fix it in a later version or at least better document this issue and provide a warning if users are using unsupported options.

grgohr, 2017-03-21:
@irtimmer thanks for the feedback
should I place some gamecontrollerdb.txt somewhere as it seems to be the SDL2 way of mapping inputs ?

irtimmer, 2017-03-27:
Yes gamecontrollerdb.txt is loaded and probably should be placed in the same directory as the moonlight binary (or maybe the current work directory)

substring, 2017-04-05:
I wouldn't mind a comand line switch to manually specify the path (+ name ?) of the gamecontrollerdb.txt ;)

Anyway, talking about SDL2, this could also be solved with #266 ... ,2 birds with 1 stone ;)

substring, 2017-04-05:
@grgohr recalbox does generate some gamecontrollerdb.txt since 4.1 because of ppsspp (sorry for the off topic comment)

grgohr, 2017-04-06:
@substring yes thanks I tried that one in all possible folders as @irtimmer proposed but still no reaction in moonlight.
It seems that the xinmo dual is not really compatible with the gamecontrollerdb mode (i had to rollback ppsspp to a pre-sdl2 version to have it work in rb4.1)

irtimmer, 2017-05-27:
Since commit 1d7c2be7e2a967f40b4a8047bf16d24c5e45dd8d you can select the gamecontrollerdb.txt file using `--mapping`. But if your controller doesn't work with SDL2 it will neither work with Moonlight Embedded.

closed 2017-05-27 by irtimmer